t-oster/VisiCut #370: Center-to-Center Engraving not working

joshikirsch, 2016-05-10:
Hey!

We are using VisiCut in our FabLab and have a rather big problem:

Whenever we try to use another orientation method than "top left", the laser still starts as if it was supposed to do so. Even our biggest nerds can't find a solution to this problem.

peteruithoven, 2016-05-10:
Hi @joshikirsch could you clarify how you change the orientation (method)? Do you mean right clicking and selecting `set startpoint`?

t-oster, 2016-05-10:
that was exactly the method I was going to suggest.

joshikirsch, 2016-05-10:
Well, after opening VisiCut with the Inkscape extension we tried to select "Center-to-Center" in the "Position" menu (between Mapping and laser-settings). The laser then still starts as if "top left" was selected.

t-oster, 2016-05-10:
This menu is to position the current drawing on the laser-bed, it has nothing to do with the laser starting point.

joshikirsch, 2016-05-10:
But it doesn't seem to change the current position of the drawing? So to select a different starting point in my drawing I'd have to right click and than define the clicked point as a new starting point? If so, is there any way to exactly set the center of the graphic as the point where the (red) laser is set?

t-oster, 2016-05-10:
It changes the reference of the coordinates. So you can set the x- and -y position of the center or corners of your drawing.

If you have an epilog lasercutter (I did not test it with any other lasercutter) you can use the x/y-off button and move the laserhead to a desired point. You can then use the right-click "set starting-point" feature to set where the laser-head's position is in the current drawing.

joshikirsch, 2016-05-10:
Thank you very much!
I was stunned by the fast replies! :D
Seems like this could solve our problem. :)

t-oster, 2016-05-10:
If it does, please close the issue.

renebohne, 2016-06-15:
is this issue solved?

closed 2016-11-17 by joshikirsch