t-oster/VisiCut #372: Cmd+W on Mac not working like in other apps

faheus, 2016-05-30:
Cmd+W normally closes the current window without quitting the application, in Visicut it does $something and brakes my workflow.

mgmax, 2016-06-01:
"$something" is very vague, could you please try to explain it further?

Do you have more than one Visicut window open? Visicut is not designed to be used like that and has a mechanism (singleinstanceport) that tries to prevent multiple instances being open. If you still manage to do so, saving settings will cause trouble because the concurrent instances might overwrite each other's settings on disk with whatever they have cached at startup. Larger work would be neccesary to make it work correctly.

What is Cmd+W supposed to do if there is only one window?

pelrun, 2016-06-02:
It's not currently one of the accelerators in the MainView, so I think $something == nop. :)

Ctrl/Cmd+W is generally identical to clicking the close window button on the titlebar. For VisiCut, that's the same as Quit.

renebohne, 2016-06-15:
So what do you expect to happen when you hit Cmd+w? Should it close all items and provide an empty sheet or should it close VisiCut?

mgmax, 2019-12-31:
Closing because there was no response

closed 2019-12-31 by mgmax