t-oster/VisiCut #374: Duplicate element then non-drag reposition gets confused

pelrun, 2016-06-29:
Was making several copies of a design by right-clicking and selecting "duplicate", then modifying the fields in the position tab. Things then got a bit screwy, with the original cut vanishing, and the copy not being selectable, and selecting via the file dropdown didn't produce the expected result either. So some internal state is getting confused.

If I duplicate the design and then drag the copy away, then everything works properly and I can reposition both copies with the position tab without any problems.

mgmax, 2020-02-19:
How to reproduce this bug:
- Load file (e.g. circle-20mm-red.svg from the examples)
- Drag to x=100mm (exact value does not matter)
- right-click
- duplicate
- do not drag, but go to the "position tab", set x=200mm, press enter key

Expected:
- first file is at x=100mm, second file at x=200mm

Actual result:
- first file disappears from the preview, but can somehow still be selected from the preview.

cdemeijer, 2020-09-28:
My first guess was that it is related to GraphicSet.clone method. I'm totally new to VisiCut and just messing with the code for now.
Although I seem to have it fixed locally, I'm not (at all) sure that I didn't break the code somewhere else.
Will clean up and test more tonight.

cdemeijer, 2020-09-30:
PR submitted. Hope that submitting a PR for the master branch is the correct way. I couldn't find the develop branch mentioned in the developers documentation :/

t-oster, 2020-09-30:
Yes, we switched to a more agile way with master being the latest version and instant release. Thanks for the PR

PR #590 (2020-09-30): Fixes #374: Duplicate element then non-drag reposition gets confused

Fixes #374 .

Holding on to the same transform reference as the GraphicSet being cloned is what caused the observed wonky behaviour.

I'm piggybacking a second commit to select the duplicated part because I feel it makes sense to match the behaviour of the 'add part' functionality.
commit 65b7f094 (2020-09-30): Merge pull request #590 from cdemeijer/master

Fixes #374: Duplicate element then non-drag reposition gets confused
closed 2020-09-30 by t-oster