t-oster/VisiCut #377: visicut marking and engraving fails after few lines

brains12345, 2016-08-16:
I have visicut running into mks sbase 1.3 using win 10. USB serial interface.
all cutting ops are fine, but after scanning a few lines the scanning stops and the serial window says loading, but by this time the software has reset.
Have tried various baud rates, different usb cables etc. without succcess. The problem would seem to be a buffer problem, but thats only my guess. I would be happy to have a go at the software myself, but as I am getting on in years (80) I would need at least a year to learn the new programming systems.
I trust this problwm can be sorted out, Thanking for your help

t-oster, 2016-08-16:
Hi, could you post a Screenshot of your lasercutter configuration in VisiCut so we can see which driver you are using and what settings.

brains12345, 2016-08-16:
Hi , Here is the seting screenshot

t-oster, 2016-08-16:
Sorry, this did not work...

brains12345, 2016-08-17:
[image]

brains12345, 2016-08-17:
carrying out further tests, I find that very small_low detail graphics can be engraved, the more detail is in the graphic, the less is engraved. I have tried different serial handshake methods but that seems not to help. the effect is like the data is sent to the board in one batch which overflows and the overflw data is lost. greetings

t-oster, 2016-08-17:
Hi, can you start VisiCut from command line and post the output (the last few lines) of an unsuccessful engraving job? You need to
* Run CMD (Win+R, type `cmd`, Return)
* Go to the VisiCut folder `cd C:\Path\To\VisiCut`
* Run via command line `java -jar VisiCut.jar`

brains12345, 2016-08-17:
Sorry must have something wrong, the command line reports 'cannot find java'

t-oster, 2016-08-17:
that is because your java directory is not in path. You need to find the java.exe file somewhere in C:\Programs\java\bin and use the full path to the exe (in "") instead of java. Something like
`"C:\Program Files (x86)\Java\jre1.8.0_101\bin\java.exe" -jar VisiCut.jar`

brains12345, 2016-08-17:
sorry I'm a dimwit, I can't find a java.exe plety of other java subfixes but not exe, please advise, and also when the faulty engraving has stopped, what do I expect to find when I enter the previous statements?
Thank you

t-oster, 2016-08-17:
Maybe Windows hides the .exe Extension,so you only See `java`. The driver should Print the whole communication so we should be able to Tell if the file is completely transferred. Just make a Screenshot of the Output when the engraving stops. 

Another way would be using Ethernet instead of serial,if your Board Supports it, this should not have the Buffer promlems

brains12345, 2016-08-18:
Ok thank you I will try further.

brains12345, 2016-08-18:
Success, I hope!
[image]
[image]

t-oster, 2016-08-18:
Yes, so it looks like the serial connection gets suddenly closed in between the transfer. Does not look like a buffer issue though, because the board always sends "OK".
Is serial connection your only possiblity or do you have ethernet?
Are you using a smoothieboard? Which version? Can you send me an example file, so I can try to reproduce the error?

brains12345, 2016-08-18:
I also not that the USB driver error record says that driver "device requires furher installation", I have no idea what that means, but it may help. The usb driver I am using is the latest I could find!

brains12345, 2016-08-18:
Ethernet is not possible due to the distance to the router !
Smoothie board is the Makerbase mks s board 1.3.
Here is the config file:

[configlatest.txt](https://github.com/t-oster/VisiCut/files/424304/configlatest.txt)
[configlatest.txt](https://github.com/t-oster/VisiCut/files/424305/configlatest.txt)

t-oster, 2016-08-18:
You could use a Crossover Ethernet cable to connect directly to the smoothieboard without a router in between. You'd have to manually configure your IPs though, so it may not be that easy to set up. However this allows you to upload your job in one step to the SD card of the smoothieboard and then execute it without further communication.

Maybe I will have time in two weeks to test on my smoothie board. I assume you tested different baud rates by changing in the smoothie-config and in VisiCut?
Does it always stop at the same point? Maybe the setting `planner_queue_size` does change something, but this is just a shot in the dark, backup the original value before experimenting.

brains12345, 2016-08-18:
Many thanks for your help, I would like to donate to your cause, please advise if this is possible. I will try a few more things and hope. I am no expert on modern programming but I don't like to give up. If I have any success I will inform you, many thanks again.

brains12345, 2016-09-08:
I guess you are very busy, ut have you any luck yet? should I be blaming the Mks board or looking for software problems please?

t-oster, 2016-09-08:
Did you have success with the sd Card approach? I would still like you to try Ethernet. You don't need a special Crossover cable, the Network Card will detect it. Just Set a fixed IP in the config file and on your Ethernet Card and try to Ping the Board. This is the easiest way to Check.

mgmax, 2016-09-08:
USB to serial connections can often be tricky. There are so many possible causes of the error that it is almost impossible to fix remotely. Do you have any other application than VisiCut to test your board via USB serial? Did you try other methods of transferring the data? Without these, I don't think anyone will be able to help.

brains12345, 2016-09-19:
Hi Thomas, having been lectured by the the Smoothie team, I think my 
problem is because I have the MKS klone board. When I have saved a bit 
of cash I will buy a genuine Smoothie board and then see what happens. 
Thank you

t-oster, 2016-09-19:
If you happen to try Ethernet or SD Card in the meantime, please report what happens. Until then, I close this issue. Feel free to reopen if you have new information.
with kind regards

closed 2016-09-19 by t-oster