t-oster/VisiCut #380: Illustrator Extension doesn't install automatically with CC 2015.3 under OS X 11.6

Pakue95, 2016-09-08:
The Illustrator extension isn't added once selected via the menu Extras>Illustrator as far as I can tell. I'm using the CC 2015.3 version on OS X 11.6. 
The AppleScript which can be found in the VisiCut package seems to work once opened manually in Illustrator, but isn't shown anywhere in the UI after "installing" via VisiCut. 
Also a little documentation on the script would be nice. As far as I can tell it just imports the currently selected objects in Illustrator into VisiCut.

Best wishes
Patrick

pelrun, 2016-09-10:
>  As far as I can tell it just imports the currently selected objects in Illustrator into VisiCut.

That is pretty much exactly what it does. Illustrator and Inkscape are much better at parsing vector files than VisiCut, so it's recommended to open files there and then use the plugin to send them over pre-digested.

mgmax, 2016-09-12:
If you could find out where and how to install the extension automatically, probably by placing the script file somewhere, then VisiCut could be fixed. Unfortunately we seem to have no developers using a Mac.

pelrun, 2016-09-12:
Every year they change the name of their product, so all the paths change, and all the automatic installers fail. Dammit, Adobe.

Maybe we should just scan for anything starting with "Adobe Illustrator" instead of fully hardcoded paths?

Pakue95, 2016-09-12:
In Illustrator CC the path is dependent on the installation language. I can give you the path for the German installation on the fly: 

/Applications/Adobe Illustrator CC 2015.3/Presets.localized/de_DE/Skripten/

It's probably a bit of a pain to implement every possible path as the folder names change with every language. I'm not sure how effective scanning is after finding the installation folder.

pelrun, 2016-09-14:
As far as I can tell from the Adobe scripting documentation, the localised folders are a non issue. They're just additional search locations, not a total replacement for the base Presets folder.