t-oster/VisiCut #381: Absolute Speed Option

koswix, 2016-09-21:
Hi,

Currently Visicut allows speed to be input as a percentage of the maximum cutting speed as set in the laser cutter preferences.

I think it would be more useful to allow speed to be entered in absolute terms - ie. in mm/s or mm/minute.

I can be engraving on one job at 400 mm/s, cutting plywood at 6 mm/s or MDF at 2 mm/s. As percentages of my max speed these start to get a bit silly and unintuitive, ranging from 100 to 0.5%, and it also makes it more difficult to translate settings between machines.

I do't know how active Visicut is anymore (I still use it every day!), but it's a change I'd love to see happen. Even if someone could point me in the direction of the files that need editing I'd be happy trying to sort it myself.

t-oster, 2016-09-22:
Hi, this highly depends on your lasercutter/board and how you configure your settings and e.g. how big your stepper motors etc are. There is no way for VisiCut to know how fast a setting is.
The epliog board e.g. accepts only a percentage of speed, which also depends on the current DPI setting.

Do you have a suggestion on how to implement that feature?

koswix, 2016-09-22:
Well that's annoying!

My use case is with the smoothieboard, where the output gcode has speed
specified in mm/min.

So it would be a trivial change (I think!) to  implement it for smoothie,
but would probably require the ui to change depending on the laser cutter
driver selected.

Maybe I could implement a preference to allow input in absolute, with a
warning that it won't work for some drivers like epilog. Selecting the
option would then bypass the percentage to speed rate conversion used in
the smoothie driver?

t-oster, 2016-09-22:
...well you just need to subclass the https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/SmoothieBoard.java class and override the `getLaserPropertyForVectorPart` method to return a LaserPropertyClass like this one https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/PowerSpeedFocusProperty.java, but with absolute speeds (just dont clamp the values to 0 and 100).

Then you would need to override the `writeVectorGCode` method similar as here https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java#L421 but with your LaserProperty implementation. Just look the GenericGcodeDriver class on how it works in general.

mgmax, 2017-01-06:
@jekhor Things do not magically start fixing themselves because someone "votes". You are welcome to contribute to the code to actually improve VisiCut. Here and in https://github.com/t-oster/LibLaserCut/issues/63 a lot of technical details and reasons were given on the implementation. If coding is out of your scope, feel free to contribute on the documentation (wiki) or new website (https://github.com/t-oster/VisiCut/issues/233). Thanks.

jekhor, 2017-01-06:
@mgmax I know, I usually send pull request instead of "voting". In this case I tried to fix this by myself but unfortunately I don't know (and don't want to learn) Java, so I cannot write any significant amount of code for this project, except of trivial fixes (this is not trivial). "Voting" may be annoying but sometimes it helps to pay more attention to things important for users. Just ignore if don't like it :)

tatarize, 2019-06-20:
I wrote my K40 driver with explicit speeds this only percents thing is actually weird. The speed ranges from so slow the head might not be moving but you can kinda hear the stepper pulses and so fast it'll rip itself up.  Even the technical recommendations give there seem to prioritize the epliog. Obviously if I can set an explicit speed and I have a min and max speed, that's significantly more information than percent because you can solve that for percent with that information but can't go the other way and solve for actual speed. Even if epilog has percent, figure out what that means in mm/s and use it that way.

matthijskooijman, 2020-08-16:
If this is implemented (which I'd like to see), for example by configuring the units to use as part of the lasercutter config, then it also means that #398 (show units in UI) must be implemented otherwise users really won't know what they're specifying.