t-oster/VisiCut #382: visicut gcode file

brains12345, 2016-09-25:
Using Visicut to create a  graphic gcode file, the file produced is 110% using pronterface, but I cant get visicut to print the gcode graphic file only a few lines.
What am I doing wrong please?

mgmax, 2016-09-25:
I'm sorry, but your report contains hardly any information to repdroduce your bug.
Please try to rewrite your report so that someone not familiar with your situation can try it out and see exactly the same error as you did. You might consider this webpage as a rough guideline to writing useful bug reports: http://www.chiark.greenend.org.uk/~sgtatham/bugs.html

brains12345, 2016-09-25:
Perhaps I could rephrase my question!
Please describe how to out put a gfile to the laser using Visicut?
thank you

mgmax, 2016-09-25:
Load your file, choose material etc. as usual.
Instead of pressing "Execute", select "File -> Export Laser code" from 
the menu. Choose the directory in which the file should be saved, enter 
a filename (e.g. "test.gcode") and press OK.

mgmax, 2016-09-25:
Sorry, I misread your question.

You can't send arbitrary gcode files from your harddisk to the 
lasercutter with Visicut. I don't understand why you want do that using 
Visicut; pronterface is the right tool for this.

Visicut is supposed to be used like this: You need to load some vector 
file (e.g. inkscape SVG) and Visicut will generate the Gcode for you and 
send it to the lasercutter. You could also save the generated Gcode to a 
file as I described before, e.g. to put it on a SD card which you later 
plug into the lasercutter.

brains12345, 2016-09-25:
Many thanks exactly what I needed to know.

t-oster, 2016-09-25:
So this can be considered as closed?

closed 2016-09-25 by t-oster