t-oster/VisiCut #383: potrace

brains12345, 2016-09-28:
Sorry but I am a dummy with some installations.
will you please describe how I can use Potrace with visicut. I have a graphic on the screen and a right click says I can vectorise it, the a window requesting Potrace installation apeers. I have downloaded Potrace but after that i am lost.
Thank you

t-oster, 2016-09-28:
Hi, which operating system are you using, where is VisiCut installed, where ist Potrace installed (and which version) and what are the contents of your PATH variable?

brains12345, 2016-09-28:
visicut is installed in programmes folder. potrace in my documents folder. Path variable I do not understand!
regards

t-oster, 2016-09-28:
so I assume windows.... could you try to copy the potrace.exe into the visicut folder?

brains12345, 2016-09-28:
will try, thank you

brains12345, 2016-09-28:
[image]
sorry That did not work!

pelrun, 2016-09-28:
I just tested putting potrace.exe and mkbitmap.exe in the Visicut folder and it found them fine.

Did you download the correct Windows version of Potrace?

brains12345, 2016-09-28:
I think so win 10 64 bit, I will check again

brains12345, 2016-09-28:
cracked it, many thanks, downloaded it from microsoft

closed 2016-09-28 by brains12345