t-oster/VisiCut #384: Feature: Origin bottom left

t-oster, 2016-10-21:
I have been asked to change VisiCut to have the origin bottom left instead of top left. I think we could make this selectable in the options. 
I would leave all processing internals as they are and only change the presentation. So it will basically use `bed_height-y` instead of `y` in many places.

Here are some mock ups, what do you think?

1. There would be a new option in the preferences
[image]

2. In the preview, the handles and the lineal would origin from bottom:
[image]
3. 
The Y field in the position panel would be seen as from bottom:
[image]

mgmax, 2016-10-22:
I personally don't need it, as the Epilog laser I am using has a physical ruler which starts at the top-left.
Nevertheless I would like to give some feedback:
- I would prefer making this a machine-specific setting, as some machines have a defined top-left and others a defined bottom-left origin (e.g. mechanical marking)
- Except for the "reference point" settings tab, it is quite clear how it works. The reference tab has been a cause of mis-understanding for a long time, because people seem to think that the dark grey square represents the whole laser bed and not the object. There should be something around it which symbolizes the laser bed and the direction of measurement.

pelrun, 2016-10-23:
Yeah, this definitely belongs with the machine config, rather than visicut.

As for the reference point issue, perhaps calling it "object reference point" instead of just "reference point" will help?

t-oster, 2016-10-28:
I did put it in the laserdevice-settings dialog. feel free to test #387

madleech, 2016-11-21:
Awesome feature! I've been inverting the measurements in my head all this time, never occurred to me it could be fixed :)

So I've tested #387, but I'm not sure it is working properly. Testing steps:
* Origin in bottom left
* Open red 20mm square and place bottom left corner at 0,0
* Generate Gcode

Result:
```
G21
G90
M3
G0 X0.000000 Y259.943600 F3600 S0
G1 X19.964400 Y259.943600 S0.200000 F500
G1 X19.964400 Y279.958800
G1 X0.000000 Y279.958800
G1 X0.000000 Y259.943600
G1 X0.000000 Y259.943600
G0 X0 Y0
M5
```

The X coordinates are fine, as expected. The Y coordinates however are unchanged. If however I use the "Flip Y Axis" option then the Y coordinates are correct.

Am I missing something or is this a purely cosmetic change?

t-oster, 2016-11-21:
As already stated `I would leave all processing internals as they are and only change the presentation`, so this behavior is intended: The change only changes the view of the coordinates in the GUI. The input and output of files should not change.

The rectangle should appear on the lasercutter exactly where it appeared before the setting, just the measurements on the y axis should start from bottom in the view.

madleech, 2016-11-21:
In that case it works as advertised :)

t-oster, 2016-11-26:
So, any objections? Otherwise i would merge this Feature into develop.

madleech, 2016-11-28:
Fine by me.

closed 2017-08-20 by t-oster