t-oster/VisiCut #386: Error: connection timed out

mdcummings, 2016-10-27:
I'm trying to drive and Epilog Helix from a Mac via ethernet. I set the IP address to the match my Epilog. Ethernet indicates that it is connect in system settings. However, when I press execute it tries and fails to connect. I've tried a regular ethernet cable and also a "crossover" ethernet cable as mentioned in the epilog manual. Can you give me any ideas about what's wrong?

t-oster, 2016-10-27:
Hi, can you ping your epilog from the command line? can you netcat to port 515? You probably need to configure your Mac to be in the same subnet as the (static?) ip of your epilog.

mdcummings, 2016-10-27:
yes, i ping 192.168.3.4 successfully (zero packet loss, 10 ms typ)
I don't know how to netcat. what's that do?
subnet mask for both is 255.255.255.0

mdcummings, 2016-10-27:
do you know anything about epilog saying I should use a "crossover" ethernet cable? Is that required?

t-oster, 2016-10-27:
if you can ping it, the ethernet cable is fine. Most current network-cards detect crossover anyway and automatically switch it, so it probably is not necessary, but it does not hurt.
You may have a firewall, which is blocking traffic to port 515? netcat is a command line utility, which just connects to tcp ports. Just open the command line and try `netcat 192.168.3.4 515` or `nc 192.168.3.4 515`. I am not sure if it is installed by default though.

mdcummings, 2016-10-28:
Thanks you for your help Thomas. I figured out my problem. I hadn't changed
"Hostname" in the manage lasercutters menu to my epilog's IP address. I had
left the default as shown below. Doh! As soon as I fixed that, it worked!
If that was documented somewhere, I must have missed it. Thanks again!
- Mike
[image: Inline image 1]

t-oster, 2016-10-28:
Please feel free to update the wiki. So we can consider this as closed?

t-oster, 2016-10-28:
...but what did you mean by `I set the IP address to the match my Epilog` if not changing the hostname?

mdcummings, 2016-10-29:
In Apple Network Settings, I set ethernet to the range 192.168.3.X.  I
thought this was all I needed to do since Ethernet showed "connected" and
ping test worked. What I didn't realize is that Visicut needed the proper
Hostname/IP for the cutter. It would be good to say in your manual, " Set
the IP address of your laser cutter in the Hostname field of Visicut's
 Manage laser cutter settings."

Also, in your downloads area, I used the older "tested" builds. I noticed
that the newer "untested" builds actually had the right default IP for
Epilog. Now I'm using your newest build. I like it!! I'd suggest you tell
newbies to use the newest builds but if they experience problems, try the
older "tested" builds. Your new builds have lots of nice features.

All the best
- Mike

closed 2016-10-31 by mdcummings