t-oster/VisiCut #388: Min Laser Power for Engraving

McNugget6750, 2016-11-06:
Hi, I'm using the Generic GCode Driver to produce engravings and vector cuts in one file.
So far it works well!

However, I'd like to be able to set the minimum power the laser needs to actually engrave something. The maximum value is already available as a setting.

Three values makes sense for calibration without the need for a map:

1. Min Laser Power (like S50)
2. Max Laser Power (like S550)
3. Off Laser Power (like S0)

The grey scale image to engrave can then be scaled to work between 50 and 550 and use 0 as the actual off value.
At the moment it is required to scale the image to work with the laser.

Best regards,
Timo

mgmax, 2017-03-12:
Are you really talking about the "engrave 3D" mode, which varies power, and not about the "engrave" mode, which uses dithering ( on/off pixels with only one level of power)? For the normal "engrave" mode this would have no effect as long as the power level you set is higher than the minimum percentage.

Even with the normal "engrave" mode, the minimum engraving intensity heavily depends on the material, and the mapping is highly nonlinear. For rather low power values you will see a change on acrylic, but nothing on wood.

mgmax, 2019-12-31:
Closing because there was no response

closed 2019-12-31 by mgmax