t-oster/VisiCut #389: Overscan for Engraving

McNugget6750, 2016-11-06:
Hi, 
An overscan option with a definable distance would make sense to get rid of the darker regions close to the border of an engraving.
Start a raster line with 0 laser power for like 3mm, start raster engraving the image, end the line with a 3mm overscan and 0 laser power.
I did this here in php: [https://github.com/McNugget6750/img2gco](https://github.com/McNugget6750/img2gco) and it works very well!

t-oster, 2016-11-06:
The Laos driver already has a setting 'Additional space per Raster line', which is probably exactly what you want. We should just port it to the gcode driver.

McNugget6750, 2016-11-06:
Sounds great! Let's do that. Can you point me to the code block? I have set it all up and just implemented the min laser power issue that I just opened. Testing it at the moment.

t-oster, 2016-11-09:
it is used here https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/LaosCutter.java#L600 but of cause the encoding here is different.

It should be added to the general convert method here: https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/LaserCutter.java#L193

but it is not so trivial, because we should use "lineto" with zero power instead of "moveto" for the white space....

McNugget6750, 2017-10-29:
After some research only almost one year later, I think I have to disagree. It should be implemented here:
LaserCutter.java:193 convertRasterizableToVectorPart()

After this runs I have no means to actually identify what the command is about. And since I know exactly what needs to happen before this method gets executed, I can easily add the overscan distance here. Even "lineTo" or "moveTo" is no problem at this point. 
The only concern is that this will affect all existing all drivers.

mgmax, 2019-12-31:
Should be fixed by #72 for all GCode-based drivers.

closed 2019-12-31 by mgmax