t-oster/VisiCut #395: Build Requirements

iceblu3710, 2016-12-03:
I get errors about missing dependencies, I add then and get some more, over and over...

Do you have a complete list of requirements for a vanilla build environment? I would like to build in a VM.

pelrun, 2016-12-04:
They should already be included in the lib directories for visicut and liblasercut.

t-oster, 2016-12-04:
If you mean stuff for building the OS specific packages,i have a docker container with everything.
I will post the Dockerfile tomorrow

iceblu3710, 2016-12-04:
A docker container would be fantastic thanks.

t-oster, 2016-12-05:
Hi,

this is a hacky version, I am using, but it works for me. Feel free to polish it and create a github project, if you think it is ready for other people to use.

It is based on ArchLinux but there are some dependencies which I needed to include separately.

[visicutbuildservice.zip](https://github.com/t-oster/VisiCut/files/630437/visicutbuildservice.zip)

mgmax, 2016-12-05:
Thomas, can you please create a Github repo anyways? Published code is always better than unpublished code.

t-oster, 2016-12-05:
here you are https://github.com/t-oster/VisicutBuilder

t-oster, 2017-12-31:
I guess this is solved by the image....

closed 2017-12-31 by t-oster