t-oster/VisiCut #399: UI: Add 'Stop Execution' button

jekhor, 2017-01-05:
There is no way for now to stop job execution if it started, need to disconnect cutter physically which is very annoying.

t-oster, 2018-10-22:
Which driver? Which device? On the Epilog devies e.g. the job is just sent to the cutter and has to be started there manually, which also works if the cable is disconnected, so there is no way to stop or start a job from VisiCut.
What are the proper commands for your device/driver to stop a running job?

darkjavi, 2018-10-24:
I understand that this is not a problem in every scenario, only tried grbl and smoothie boards but as far as I've seen there are to different workflows, in Epilog/smoothie(in html mode) the job is sent to the machine as a whole and then you can start/stop/pause the job with the machine interface, this makes perfect sense.

On grbl/smoothie( ip or serial mode) visicut acts as a host and streams the job to the machine, problem there is that once you press execute you lose control over the proccess so if you need to stop you have to close visicut || pull usb || kill switch. Main thing is the annoyance of having to set it up again for the smallest mistake like realizing that you need 10% more laser in the first 5 seconds.

For those cases would be great to be able to pause/stop the stream, maybe a little trickier cos you need to make sure laser goes off.

t-oster, 2018-10-24:
Yes, I understand the issue. Currently I am not using any `streaming` Lasercutter so pull-requests are appreciated. I'd suggest an Interface `StoppableLasercutter` which those drivers should implement which provides the stop method. Additionally there should be a config value, because some drivers allow both modes.

mgmax, 2018-10-24:
AFAIK all Most of the streaming drivers are blocking, e.g.

mgmax, 2018-10-24:
AFAIK all of the streaming drivers are blocking, e.g. sendJob() in GenericGCodeDriver does not return until the whole job is sent.
https://github.com/t-oster/LibLaserCut/blob/d07e53259a96912dbf435bc045a8695eef08d8ce/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java#L790

My suggestion would be to
1. call sendJob from a Thread  (this is already done!)
2. add a cancel button that uses Thread.interrupt() to signal the thread that sending should stop
3. use Thread.interrupted()  to check for that inside the loop in writeVectorGCode(). From there, throw `InterruptedException` to get back out. Catch it in sendJob, make sure to send some end-of-job commands (switch off laser etc.) and close the output device.

Just to be clear, same as @t-oster I won't be the one who implements this, I can just help you do it.

For inspiration, something similar is in the engrave dithering code to abort calculating the preview:
https://github.com/t-oster/LibLaserCut/blob/d07e53259a96912dbf435bc045a8695eef08d8ce/src/com/t_oster/liblasercut/dithering/Ordered.java#L88-L90
https://github.com/t-oster/VisiCut/blob/af8129d9a46fc7ca15ef51a57dbeab3f25de6510/src/com/t_oster/visicut/gui/beans/PreviewPanel.java#L290-L295

And the best thing is that the first step is already done: sendJob is called here, and this is already inside a thread. https://github.com/t-oster/VisiCut/blob/0beb401ea0c0f78961e9f1db86fc8cab7cf5524e/src/com/t_oster/visicut/gui/MainView.java#L2255
But now it's up to you @darkjavi .

tatarize, 2019-06-19:
The K40 can do this. The job streams to the device without the entire queue posted, but even when the queue is in there it can be sent a `IS1P` command mid cut and will clear the queue. It also possesses a few other things that are common for moving the laserhead around before starting the project. So you can be absolutely sure where you're going to cut. And generally just call that start point the origin point. Also a couple other helpful commands like unlocking the rail so you can move the head around yourself, firing the laser without anything moving (this one actually scares me but it'll do the same as a test button). There were no clear places to tie these elements in, but they are functionality possessed by the devices.

mgmax, 2020-08-16:
Related: https://github.com/t-oster/LibLaserCut/issues/158

TheAssassin, 2024-03-07:
Related: #706