t-oster/VisiCut #400: Segmentation fault with openjdk-9

jnweiger, 2017-01-15:
I am getting a crash using either your https://github.com/t-oster/VisiCut/releases/download/1.7_310/visicut_1.7-310-gcf8c087-1_all.deb or my own build https://github.com/fablabnbg/VisiCut/releases/download/1.7.0%2B20161220jw/visicut_1.7.0.20161220jw-1_all.deb

The splash screen shows for a second, then disappears:
<pre>
$ visicut Eiffelturm-jw.svg 
Can't load settings.
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007ffb332be009, pid=24050, tid=24074
#
# JRE version: OpenJDK Runtime Environment (9.0) (build 9-internal+0-2016-04-14-195246.buildd.src)
# Java VM: OpenJDK 64-Bit Server VM (9-internal+0-2016-04-14-195246.buildd.src, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libjava.so+0x1d009]  JNU_GetEnv+0x19
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/jw/fabcloudnbg/jw/objects/hs_err_pid24050.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
/usr/bin/visicut: Zeile 10: 24050 Abgebrochen             java -Xms256m -Xmx1024m -jar "$DIR/Visicut.jar" --singleinstanceport $port "$@"

</pre>

jnweiger, 2017-01-15:
[hs_err_pid24135.log.txt](https://github.com/t-oster/VisiCut/files/706892/hs_err_pid24135.log.txt)

jnweiger, 2017-01-15:
I can switch to openjdk-8 via update-alternatives, then it runs repeating the following warnings:
<pre>
WARNUNG: Could not load font 'sans-serif'
</pre>
As the crash is from within libfreetype, I'd suspect something font related.

t-oster, 2019-11-05:
since we agreed to support only java 11 from now on, I think this can be closed. Or is there any usecase for java 9?

closed 2019-11-07 by mgmax