t-oster/VisiCut #401: "INNER-FIRST" Causes  strategy generates incorrect G-code

Walkingrobot, 2017-01-23:
When using the "INNER-FIRST" strategy on some large .svg files some lines are not converted to G-code.  The "NEAREST" is working fine with the same file. (Using the inkscape plugin or by importing)

You can find the setting for visicut and smoothieboard here => [https://bitbucket.org/dashboard/overview]

Will do some more testing with other source files.

mgmax, 2017-01-23:
The link to your config seems broken. Please also try to find a short example file which shows the problem.

The relevant code (written by me) is in LibLaserCut:
https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/vectoroptimizers/InnerFirstVectorOptimizer.java
Please open a bug report in the LibLaserCut project.

As it only uses `Collections.sort()` on the list of paths, I cannot imagine how lines would get lost. The algorithm doesn't sort nicely if the paths are not connected, i.e., split apart into many fragments, as noted here: https://github.com/t-oster/VisiCut/issues/203
It might be that something in the G-Code postprocessing chain then throws away these short fragments, although I don't know what that could be.

Walkingrobot, 2017-01-24:
Did some more testing today.  Issue is caused by Smoothieboards upload function. (Corrupted file)  Nothing to do with Visicut.  Coincidence it was working with the "NEAREST" strategy. 

Best regards

closed 2017-01-24 by Walkingrobot