t-oster/VisiCut #402: How to localize

obujor, 2017-02-02:
Hi,

thank you very much for this software! 

Here @WakenMakeFabLab we would like to use VisiCut but not all of us know English, so we need the Italian interface.

Could you please explain how to add a new language to the interface?

Thanks
Octavian

pelrun, 2017-02-03:
Have a look in the resources folders, for instance https://github.com/t-oster/VisiCut/tree/develop/src/com/t_oster/visicut/gui/resources (though there are others in the tree.) These contain .properties files which hold the text for the various languages (english is the default, obviously.)

obujor, 2017-02-03:
Thanks @perlun, where can I choose which language to use in the gui?

pelrun, 2017-02-03:
It's not directly selectable; instead it pulls the locale from your system. So if your OS is already in Italian and that translation is available in VisiCut, then it'll be automatically selected.

obujor, 2017-02-03:
Hmm, ok.
It's cool that it gets the locale from the system but in my opinion a selectable option is a nice to have feature. For example my OS is in English but if I want to show someone how to use VisiCut I would like to have the UI in Italian and after switch back to English.

Thank you a lot @pelrun for the explanation.

closed 2017-02-03 by obujor

t-oster, 2017-02-03:
You should be able to override the settings if you want, by giving the variables to java on startup:
`java -Duser.language=th -Duser.country=TH -Duser.variant=TH -jar dist/Visicut.jar`

found here http://stackoverflow.com/questions/64038/setting-java-locale-settings

obujor, 2017-02-03:
Thanks @t-oster good to know!