t-oster/VisiCut #403: Engraving: Exporting of Laser Code(G-code) does not work when rastering

Walkingrobot, 2017-02-19:
I'm having troubles exporting g-code when engraving a job. (File -> export laser code)  File is empty.  If I use the execute function it does work.

t-oster, 2017-02-23:
Hi. I tried it with the latest version and exporting of raster-code seems to work. But I tried it with a very small file and it took quite some time. I will look into it, maybe it is too slow for big files. Please try with a small rectangle.

Walkingrobot, 2017-02-24:
I used the 20mm rectangle example.  Same result.

**Tested on:**  Windows 10, Debian Wheezy
**Lasercutter driver:** Smoothieware
**Visicut version:**  1.7-310
**Workflow:**
1.) File -> Example -> 20mm_rectangle
2.) Mapping -> engrave everything (Floyd-Steinberg)
3.) Power 20% en speed 100%
4.) File -> Export laser code => This result in gcode file no reastering G-code, only startup g-codes.

Also tried importing the settings from Berlin fablab, same result.

[test.txt](https://github.com/t-oster/VisiCut/files/798145/test.txt)

Could it be a driver problem ?

Thanks !

t-oster, 2017-02-24:
Is there any console output if you start visicut from command line?

Iliasb, 2017-02-27:
Nothing special:  Only this at startup =>

VisiCut.Linux: 1: VisiCut.Linux: Bad substitution
VisiCut.Linux: 9: VisiCut.Linux: let: not found

t-oster, 2017-02-27:
What is 1.7-310? Should be something like 1.7.9-6x please try with the latest version from http://download.visicut.org/develop

Walkingrobot, 2017-02-28:
Fixed in version 1.7.9-6x (Develop branch)
Broken in 1.7.310 (Master) ;)

Thanks

closed 2017-02-28 by Walkingrobot