t-oster/VisiCut #404: New Laser support for Bodor BCL 1309-XM

jnweiger, 2017-03-02:
Hey all!
I'd like to collect any info about this laser cutter: http://www.bodor.com/CO2_Laser/87.html
Do we know if it is already (partially) supported by one of our driver backends? Or if it is similar to one?

This is what I have so far

* RUIDA Board
* (W)LAN and USB support

jnweiger, 2017-03-02:
* Example hexdumps generated for another laser with a RUIDA board:

[test.rd.xxd.txt](https://github.com/t-oster/VisiCut/files/814650/test.rd.xxd.txt)
[Rechteck.rd.xxd.txt](https://github.com/t-oster/VisiCut/files/814651/Rechteck.rd.xxd.txt)
[Rechteck_klein.rd.xxd.txt](https://github.com/t-oster/VisiCut/files/814648/Rechteck_klein.rd.xxd.txt)
[Rechteck_3seiten.rd.xxd.txt](https://github.com/t-oster/VisiCut/files/814649/Rechteck_3seiten.rd.xxd.txt)

t-oster, 2017-03-02:
Does not look very human readable/text based. IMHO all things we already implemented are ASCII, so your best shot is the leetro thing #159

jnweiger, 2017-03-02:
Thanks for the hint!
The MPC6525-Controller-Mainboard and RUIDA Board look very different from the outside.

pelrun, 2017-03-03:
It's RDWorks, yaaaaaaay. :|

The files are (weakly) scrambled; the encryption and a lot of the format messages are reverse engineered here: http://stefan.schuermans.info/rdcam/

There are more message types in the example files given than are documented, but it looks pretty feasible to write a plugin for this.

jnweiger, 2017-12-09:
We have sucessfully decoded the RDWorks file format. There is now a new inkscape extension (written in python) that does the job. https://github.com/jnweiger/inkscape-thunderlaser/releases

Next step is porting the code to Java. 

Help welcome!

kkaempf, 2018-01-07:
https://github.com/kkaempf/LibLaserCut/releases/tag/v0.0.4 fixes this :wink:

pelrun, 2018-01-07:
@kkaempf I don't think that's the right way to go about that change, as you've added "min power" to every driver even though it's not appropriate. 
You should make a *new* subclass of LaserProperty with the fields your driver needs, then override the getLaserPropertyForVectorPart/getLaserPropertyForRasterPart methods in the driver to return an instance of that class.

kkaempf, 2018-01-07:
@pelrun thanks for the review ! Await my pull request :wink:

jnweiger, 2018-01-10:
I've made a release out of that branch and tested the code. Seems to work fine!

https://github.com/fablabnbg/VisiCut/releases

mgmax, 2018-12-26:
@jnweiger Could you merge your changes into LibLasercut (maybe label the driver as experimental)? I fear that the code will diverge too much and it will become difficult to merge it later. (Currently, I'd guess you only need to care about https://github.com/t-oster/LibLaserCut/commit/087c2667002d8ffcd49833c4ebbef2f4b2520dfe , but more changes may come in the future ).

t-oster, 2019-03-02:
@jnweiger what is the state of your driver? Wouldn't it be better to have just one VisiCut version with all features? (We have proper OSX packaging now ;)

jnweiger, 2019-03-02:
I agree, we should merge asap. I was not able to spend more time on Visicut unfortunately.
There is also work by @kkaempf and @mrlinux1 which I apparently had failed to merge correctly into my branch. I am seeing bugs that are supposed to be fixed already :-(

mgmax, 2021-01-07:
Closing this issue in favor of https://github.com/t-oster/LibLaserCut/issues/105

closed 2021-01-07 by mgmax

kkaempf, 2022-05-21:
This should settle it: https://github.com/t-oster/LibLaserCut/pull/180