t-oster/VisiCut #405: Is the UTF-8 encoding supported?

Richmond96, 2017-03-05:
Hi,

I'm localizing to Italian your software for @WakenMakeFabLab. When I launch your program it looks like the UTF-8 encoding isn't supported.

Is that right or ther's a way to enabling UTF-8 encodign?

Thanks
Riccardo

t-oster, 2017-03-06:
well IMHO java and swing should be UTF-8 by default. How are you making the translation? Are you following https://github.com/t-oster/VisiCut/wiki/How-to-translate-VisiCut ?

Richmond96, 2017-03-06:
Yeah, I'm following your instructions, but I launch the project with Netbeans and I notice that there's a problem with accented characters. 
I also checked that the UTF-8 encoding was enable for the project and I've tried with other project and they didn't have any problem.
When I launch yours, the display of accented characters is incorrect

t-oster, 2017-03-06:
Could you post the utf8 encoded Translation Files, so I can Check this?

Richmond96, 2017-03-06:
Sorry, I used the native form (à,è,ì,ò,ù) to insert the accented characters.
Instead using the corresponding ascii code there no display problem for the accented characters

closed 2017-03-06 by Richmond96