t-oster/VisiCut #413: No Camera Image shown from visicam

domonoky, 2017-04-26:
I tried visicut 1.7.9 on windows 7 and on Debian Linux (Raspian) and it does not show the image from the webcam (visicam) on the (virtual) laser bed.
But the camera itself works, i get an image if i go to the calibrate camera page.
Also if i use the right mouse button menu to copy part of the camera image, i get the selected part of the image shown. But the virtual laser bed stays grey.

Anyone an idea on how i could debug that issue ?

mgmax, 2017-05-14:
Sounds strange. Did you enable the camera in the menu? Otherwise the image is not shown.
Did you also calibrate the camera?

Nos-, 2017-08-07:
Yes, on my first PC (with external WebCam) it worked fine but on my Notebook it doesn't work. It seams that the numbering is important (does it start with 0 or 1).

mgmax, 2017-08-07:
I don't understand how your comment is related to the original one. Did you set the correct OpenCV device number? Does the VisiCam log show an error message? Can you view the webcam using another tool like vlc?

mgmax, 2018-05-05:
I found part of the cause of this bug.
Steps to reproduce:
- Add a new lasercutter
- another bug: don't set an empty name or VisiCut will die a horrible death
- Set this camera URL (random URL from the internet) http://www.citypictures.net/data/media/136/Taiti_wallpaper_1600x1200.JPG
- save
- Now only a light gray background is shown, independent of the material color.
- Right-click and copy camera image -> image becomes visible!

Calibrate the camera:
- Open the calibration dialog.
- Another bug: The markers are not shown! Change the number of markers.
- ... do the calibration

Then everything will be fine again.

However, there must be some race-condition(?) that kills the calibration settings, which is most often triggered when downloading settings for the first time or shortly after.

PR #456 (2018-05-06): Fix Bugs in Camera Calibration and Manage Lasercutters dialog
+ fix behaviour if camera not calibrated
+ add some translations

Fixes #413 
(please open a new bugreport for the suspected remaining bug which kills the camera calibration)

I tested it on Linux until I couldn't find further bugs in the code I added.

lesson learned: New cool features like the many-point camera calibration have a significant QA overhead, no matter how well the code was written.
commit 64fa9e3e (2018-05-06): Fix Bugs in Camera Calibration and Manage Lasercutters dialog

+ fix behaviour if camera not calibrated
+ add some translations

Fixes #413
(please open a new bugreport for the suspected remaining bug which kills the camera calibration)
closed 2018-05-06 by t-oster