t-oster/VisiCut #414: Problem with long lines meeting circles

thatch, 2017-05-08:
I understand this is a very high-level bug report; I hope to make the repro simpler and narrow down whether it's Visicut's generation or the laser's interpretation.

I cut the attached dxf using VisiCut 1.7.9-72-g4f41f657 on an Epilog Mini (using the Helix profile).  On load it complains about not being on the bed; I dragged it another 3mm away from the origin to make sure.

When cutting the outline, the head proceeded counterclockwise.  The end of the long angled lines were slightly too far down (what the preview calls +y) -- about 0.5mm.  The machine then makes a short whine, before starting the arc in what I assume is the correct place.

I didn't get a very good photo, but this is the topmost (closest to y origin) part of the cut. 
 https://goo.gl/photos/PXjhHiJ1KTPXGQ3i9

[visicut_bug.zip](https://github.com/t-oster/VisiCut/files/984618/visicut_bug.zip)

thatch, 2017-05-09:
I moved the part around in preview and managed to make it go clockwise; the error moved, and continues to happen at the end of the long lines.  Comparing the two pieces, the holes and outer arcs line up perfectly, but the lines don't.

thatch, 2017-05-09:
Here it is rendered with ghostpcl; that looks fine to me.  I'll follow up with Epilog.

[output.pdf](https://github.com/t-oster/VisiCut/files/987781/output.pdf)

mgmax, 2017-05-14:
As this is not a bug in VisiCut, could you please mark it as "closed"?
For following up with Epilog, I would suggest creating a simple file in Corel Draw which triggers the bug when using the official Epilog driver. Otherwise they probably won't care.

mgmax, 2017-08-07:
@thatch Could you please close this issue if it is not a bug in VisiCut, but the Epilog Firmware?

thatch, 2017-09-12:
I assume it's error accumulation due to fixed-point math in the Epilog firmware; we could work around it in visicut by using shorter line segments, but I will close for now.

closed 2017-09-12 by thatch