t-oster/VisiCut #416: optimize order of vectors to process

Nos-, 2017-08-07:
If I want cut some contours it should be done in an efficient way. E.g.: cut one vector-curve and afterwards take one which begin is located in the near. But currently there will be cutted something in one area and afterwards something in an totally different area and afterwards cutting in the first area will be continued which is totally inefficient.

The reason seams to be that vectors send from Inkscape to VisiCut in the order of their last modification and not based on location. Maybe an nearest-neighbor sorting would be helpful.

mgmax, 2017-08-07:
This is possible by choosing the sorting strategy in the laser profile, although not really documented:
https://github.com/t-oster/VisiCut/wiki/Documentation:-Sorting-strategies
https://github.com/t-oster/VisiCut/wiki/VisiCut-manual

By the way, if something does not work as expected (such as the inkscape plugin), please do not write the workaround into the user manual, but create an issue here.

mgmax, 2018-01-13:
@Nos- Am I right that the issue is solved for you? Please close it.

mgmax, 2018-08-19:
@t-oster Could you please close this issue?
(P.S. There are about a dozen other issues which should be closed IMO because there's nothing we can do if the original author doesn't respond, or because they are 99% fixed and the original discussion has become irrelevant. Is there a better way instead of writing these "please close" comments?)

closed 2018-08-19 by t-oster