t-oster/VisiCut #417: Feature request - True greyscale raster engraving

funinthefalls, 2017-08-16:
What are the chances of having greyscale instead of dithered output for raster engraving?

mgmax, 2017-08-16:
We already have that for Epilog lasers, it's called 3D engrave.

funinthefalls, 2017-08-16:
Will this work with the GRBL driver as well, or is it specific to the Epilog driver?
(BTW, thanks for the lightning fast response to my question)

pelrun, 2017-08-17:
The rasterizer is driver-independent, so Raster 3D is definitely supported by all the GCode driver variants.

t-oster, 2017-08-17:
This is not correct....

t-oster, 2017-08-17:
This is correct, but the Epilog has special support for Raster 3D while the other drivers use a `raster(3d) to vector conversion` which is not optimal.

pelrun, 2017-08-17:
It's not like the Epilog doesn't have to do that internally anyway.

t-oster, 2017-08-17:
In some whay of cause, but this does not influence the file-transfer time. Also I think epilog uses some advanced motion-control and the head runs much faster in engraving-mode than in vector mode, because it is limited to the x-axis. Also I am not sure if the GRBL/GCode devices really travel with a constant speed when the `line` consists of many `G1`s with different laser intensities.

pelrun, 2017-08-17:
The GCode engines generally do forward motion planning, so there shouldn't be any issues there. I certainly don't see any on my SmoothieBoard. Grbl used to have an issue with stopping at speed changes (because it believed it was driving a spindle) but the latest version has a laser mode specifically to fix this.

funinthefalls, 2017-08-17:
Thanks for the information, I will close this issue now.

closed 2017-08-17 by funinthefalls