t-oster/VisiCut #418: Use Project Lombok to make the code more readable

t-oster, 2017-08-17:
When I look to the code I see most of it are Getters and Setters which distract from the real logic and makes the files very long.
In my newer projects I use https://projectlombok.org/ which allows to annotate variables with `@Getter,@Setter` or even wohle classes with `@Data` which results in the getters and setters to be auto-generated.
It is supported by the most IDEs out of the box, however it is still somewhat **non-standard**.
In my opinion the advantages are more, but what do you guys think?

phdoerfler, 2020-03-11:
Yes please!

t-oster, 2020-03-11:
Well there are also some Drawbacks: Some Lombok/Netbeans combinations make problems and for eclipse and others you have to install it first....

phdoerfler, 2020-03-11:
That wouldn't stop me. But then again, I'd just switch to [Scala](https://www.scala-lang.org) instead.

t-oster, 2020-03-11:
If i would switch, then Kotlin would be my first choice. However, since the NetBeans Plugin for Kotlin is not ready yet, I stick to java