t-oster/VisiCut #420: [Feature Request] Language selection

clemclemmm, 2017-10-05:
there is more than just the english version of the software, 
(even if some of them are partially translated)
but no language selection in settings, 
i don't know if it's easy or not to add this feature. 

but it could look like : 

Language : 
- English
- German 
- French
- Italian
- Dutch
- Translate It Yourself and add your own translation --> send to the appropriate wiki page

subtext : all the language are community translated, so if one is incomplete, you can do it by yourself, and send it to us.

i have translate to french today, looking now how to send it back to the github :-)

clemclemmm, 2017-10-05:
i don't really understand yet how github work, i'm not a dev, i hope i done it right. 
at least, i have tried.

mgmax, 2017-10-06:
Hi, thanks for contributing!
VisiCut automatically chooses the language depending on your system settings, for example the environment variables on Linux systems. For example, on Linux and Windows machines I use it is automatically in German. Does that not work for you? Do you still see the need for a configuration dialog if it works?

clemclemmm, 2017-10-06:
I have publish the files, but i'm not using it yet for myself, on my computer.
I will try.
But if automatic, it's better than a dialogue selection menu, i guess :)
So : thanks!

ClemClem

clemclemmm, 2017-10-07:
it seems that i dont know how to integrate that french langage fail on my OSX app. (if i open the container, there is no directory for adding these files. 
and i dont know how to compile it on my computer (because i think it's what i have to do, no ?)

mgmax, 2017-10-07:
The files are compiled in, you would have to build the sources using NetBeans, see https://github.com/t-oster/VisiCut/wiki/Development:-Getting-started .
Or just relax and wait for Thomas to merge your changes, so that the next development version also speaks french  :-)

t-oster, 2017-10-08:
If it does not work with the latest version, please reopen.

closed 2017-10-08 by t-oster