t-oster/VisiCut #424: Cutting problem

cklam26, 2017-10-16:
Hi
I used visicut yesterday on my diy machine based on marlin firmware (I just finished it) . I tried the box example with the cut profile.
I realized by analyzing the gcode, that visicut does not say to stop the laser during non-cutting travels.
Because of this, my white paper of test shows a cut through the pieces.
This is normal in this profile or I haven't configured well visicut ?
Sincerly

pelrun, 2017-10-17:
The convention is that the laser is off during G0 moves. Marlin isn't intended to drive a laser-cutter, so it's not handling this correctly. You need to use a laser-cutter specific firmware - either a Marlin fork, grbl, or something else.

cklam26, 2017-10-17:
So G0 is for non-cutting travel and G1 is for cutting travel ? If it work like that, i'll make my own modifications on marlin firmware. Normally, the latest marlin firmware can be configured to using laser engraver (or spindle).  
If the G0 move turn off the laser, why two M5 command  are here in the generated gcode ?(At the begining, and at the end ). What is the difference ?

pelrun, 2017-10-17:
That M5 is hardcoded into the Marlin driver for some reason (alongside a homing command), probably because of the marlin fork used by the person who wrote the driver. It's the "spindle off" command, so it's probably there to ensure the laser is in a safe/known state at the start/end of the job.

mgmax, 2019-12-31:
I assume the issue is solved. If there is still a remaining issue, please answer.

closed 2019-12-31 by mgmax