t-oster/VisiCut #430: text in splash screen broken

mgmax, 2017-12-27:
Something is missing on the official buildserver: The text in the splash screen is only shown as squares, like □□□□□□□ □□□.

t-oster, 2018-01-15:
hmm... i guess it's missing a font in the https://gitlab.com/t-oster/visicutbuildservice docker-container... I will look into it.

renebohne, 2018-04-29:
The splash screen is still broken, tested in 1.8.23 screenshot attached.
[image]

mgmax, 2018-04-29:
@t-oster could you try to add the ttf-freefont package to your Dockerfile?

t-oster, 2018-04-30:
well.. if I rebuild the container with the latest arch-linux baseimage, it does not work anymore. I guess the checkinstall command has to be updated... 
Is this a wild guess or can you confirm that the ttf-freefont package would solve it so it is worth fixing?

closed 2018-05-19 by t-oster

t-oster, 2018-05-20:
fixed thanks to @mgmax