t-oster/VisiCut #433: Is VisiCut (and LibLaserCut) compatible with laser cutting machine Epilog FUSION?

drdavide, 2018-01-18:
I have read that VisiCut is compatible with Epilog ZING, MINI and HELIX. I want to use LibLaserCut for one of my project. Can I control also Epilog FUSION with VisiCut?

mgmax, 2018-01-20:
Guessing from the Epilog driver and Job Manager output, and some network traces I could get from a test device, I can say the following:
The Fusion's command set is slightly different from the Helix. Slight modifications to the VisiCut code will be necessary. The eView integrated camera system will be possible somewhat tricky to implement with VisiCam, as it consists of two cameras and has some stupid network protocol.
Here are some scripts I wrote to get this information, and also further notes: https://github.com/mgmax/epilog-fusion-dataformat

I will definitely make all this work if/when we get a Epilog Fusion, which will be within this year with high (but not 100%) probability.

Nos-, 2018-03-19:
Hi there.

I am using VisiCut with an Epilog Legend LaserCutter - many thanks for making VisiCut by the way.
Now I can make some tests on an Epilog Fusion too. But how to start?
Is the Fusion-series already supported?

mgmax, 2018-03-19:
Start at the Helix or Legend code (whichever supports autofocus) and compare the output data of the Windows driver (print to file) with that of VisiCut. Then change the VisiCut output (Export Laser file) until it works (in doubt, until it is identical to what the Windows driver creates).

Some notes: https://github.com/mgmax/epilog-fusion-dataformat/blob/master/NOTES.txt
dump raw print to somewhat readable format: https://github.com/mgmax/epilog-fusion-dataformat/blob/master/dump-epilog-raw-print-file.py

mgmax, 2018-09-23:
As my lab has decided to not buy an Epilog Fusion, I will not work on a driver for this machine.
First, you will need to make all small details work which are different from other machines (focus, specifying speed and power, ...).
Then the main chunk of work will be implementing spline support in LibLaserCut:
- rewrite ShapeConverter to pass on splines (currently, there the path is converted to polyline with lots of LINETO commands)
- add some fallback option so that a driver still receives a polyline if it doesn't understand splines
- use the splines in the Fusion driver and send them to the lasercutter