t-oster/VisiCut #437: Timeout error exception

zebulong, 2018-02-01:
Hello-  when I am trying to send a marking job to the laser I sometimes get "Error  Exception: Timeout" and the job does not send.  It happens after about 10 seconds of waiting for the job to send.  Normally jobs go thru in just a few seconds.   anyone have an idea of what could be causing this?  using Visicut 1.8-5 on a Mac and 9 times out of ten the jobs go thru fine.  Currently I am trying to send a vector that is not overly complicated or big.  thanks for your help!

mgmax, 2018-02-01:
Which lasercutter driver are you using?

zebulong, 2018-02-01:
HELIX by Epilog
Is that what you mean?

zebulong, 2018-02-28:
The most current version is doing this as well.  Not a complex file I am giving it, but it just hangs and the progress bar sticks and then it says TIMEOUT.   any idea of how to proceed?  I have tried re-saving the file with other programs, still, the timeout happens.

t-oster, 2018-02-28:
Hi, can you rule out network-issues? Is the same file always not working or just one of ten times? Please try starting VisiCut from the command line in order to get a more useful error message.

zebulong, 2018-02-28:
I think it must be a network or similar issue, and it is one in ten.  i restarted the application and it ran OK, but it did have some re-sizing issues at first.  I was not initially able to grab the corner to scale it smaller, but if i grabbed the top middle tab and moved it first, then the corner would move.  The timeout error happens in one in ten, like you suggest.    I do not know how to start it from the command line- i know about how to access the command line, but what do i type there to start the app?   (sorry i am not an expert in computer stuff!)

james

mgmax, 2019-01-03:
I still suspect it's a network issue if it works 9 of 10 times, because I can't think of anything inside VisiCut which can cause this kind of problem.

This is how you can run VisiCut from the command line to get detailed error messages:
https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline

Does it work with the Epilog Windows driver?

zebulong, 2019-01-07:
It does it every time.  I have tried restarting the laser, downloading other versions of Visicut, etc.

mgmax, 2019-01-08:
What do you mean by "it does it every time"?
Does it work with the Epilog Windows driver over the same network connection?

Please note that the lasercutter does not support USB and Ethernet at the same time, you need to restart it when switching between USB and Ethernet connection.

zebulong, 2019-01-08:
I am using a Mac, no windows machine here.  This is the reason I use Visicut, because if I used a windows machine I’d used the Epilog driver.
 Every time I try to place the start point somewhere inside the drawing, the laser runs into an invisible wall in the software and it won’t cut to the left of a certain point, out in the middle of the laser bed.  When I cut using regular start point in the corner, it works fine.    Does this make sense?  I use only ethernet, never USB.  Should I try this using USB, is that possibly a solution?

Thanks for your help!

zebulong, 2019-01-08:
> Ah okay, so you were not referring to the network problem with "every time". Then just ignore my comments on that. My question about the Windows driver was to find out whether the problem is really with VisiCut. Even with the Windows driver I have experienced sporadic network upsets from time to time, so it may or may not be a problem outside of VisiCut.
>
> Does restarting VisiCut after every sending help? Can you use a network packet sniffer to record what's going on on the network (close all other applications during that) if the problem occurs? For Linux I would use Wireshark, but I have no idea about MacOS, Google says that Wireshark for MacOS is broken.
>
> Please discuss the other issue in #496 <https://github.com/t-oster/VisiCut/issues/496> ("starting point, invisible wall") and not here, so that things don't get mixed up. Please use this discussion here only for the "network timeout" issue. I understand your problem in both issues.

Oh, I am very sorry, I thought we were talking about the 496 issue here.  I didn’t read the subject line, because this issue is not very much on my mind, as it does not happen every time.  I see it every once in awhile.  The new issues I have are the latest downloads will not even open, even with gatekeeper released.   I posted that issue separately the other day.   Thanks for your help

t-oster, 2019-01-16:
So this can be closed, because it seems to be a network error on your setup? If not, feel free to reopen.

anjiro, 2019-02-27:
I'm also experiencing the Timeout Error exception. My setup is as follows:

- Epilog Helix 60W 24x18
- Connected via Ethernet directly to Mac running OSX 10.14.3
- IP address of laser cutter: 192.168.1.44
- IP address of Mac: 192.168.1.40
- VisiCut 1.8-62-g960eda75 (version with Java 8 included from https://download.visicut.org/VisiCut.app.zip)

I can ping the laser cutter and I can telnet to 192.168.1.44 port 515. However, when I set up a file and hit Execute, I get the `Exception: Timeout` message. Running from the console via `VisiCut.app/Contents/MacOS/JavaAppLauncher` I get the following:

```
java.lang.Exception: Timeout
	at com.t_oster.liblasercut.drivers.EpilogCutter.waitForResponse(EpilogCutter.java:156)
	at com.t_oster.liblasercut.drivers.EpilogCutter.waitForResponse(EpilogCutter.java:125)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendPjlJob(EpilogCutter.java:226)
	at com.t_oster.liblasercut.drivers.EpilogCutter.realSendJob(EpilogCutter.java:342)
	at com.t_oster.liblasercut.drivers.EpilogHelix.realSendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendJob(EpilogCutter.java:400)
	at com.t_oster.liblasercut.drivers.EpilogHelix.sendJob(EpilogHelix.java:35)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:786)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2275)
java.lang.Exception: Timeout
	at com.t_oster.liblasercut.drivers.EpilogCutter.waitForResponse(EpilogCutter.java:156)
	at com.t_oster.liblasercut.drivers.EpilogCutter.waitForResponse(EpilogCutter.java:125)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendPjlJob(EpilogCutter.java:226)
	at com.t_oster.liblasercut.drivers.EpilogCutter.realSendJob(EpilogCutter.java:342)
	at com.t_oster.liblasercut.drivers.EpilogHelix.realSendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendJob(EpilogCutter.java:400)
	at com.t_oster.liblasercut.drivers.EpilogHelix.sendJob(EpilogHelix.java:35)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:786)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2275)
```
which unfortunately does not seem particularly helpful.

t-oster, 2019-02-27:
Well... it is helpful.. It says that here: https://github.com/t-oster/LibLaserCut/blob/master/src/com/t_oster/liblasercut/drivers/EpilogCutter.java#L226 after VisiCUt sends the `\002\n` command to the lasercutter, it does not respond in three seconds.
Can you print on the lasercutter from Windows? Maybe they changed something in the firmware...

Does it happen everytime or are you sometimes able to successfully send a job?

anjiro, 2019-03-15:
It seems that the laser cutter got a loose cable on the inside, so it just wasn't responding at all to any requests. I might suggest a more informative error message, but now things are working properly.

t-oster, 2019-03-15:
Then I assume it's fixed

zebulong, 2019-04-06:
The issue seems to be alive still.  I just opened the most recent Visicut and loaded a drawing to mark.  I set some speed and power levels and sent it to the laser and it successfully ran.  It was cutting the wood too deeply, so I came back to Visicut and only adjusted the speed, from 30 to 40. Then I executed it and got the Error: Exception Timeout notice after it sits.  I closed the program, reloaded it, same thing.  I restarted the laser, replugged the ethernet and re-opened Visicut.  Now it is nothing but Exception Timeout error code, when I run this drawing.  If I close it and open something else, something simple, it will execute it fine.  But if I try to reload the original it gives the error.  But it ran once successfully already, but no longer!!????   it's strange.  any idea what it could be?   thanks for your time!

zebulong, 2019-04-06:
I posted something on this, because it seems still active.

mgmax, 2019-05-05:
@t-oster can you please reopen the issue?

@zebulong The cutting speed definitely has no influence on connection/network errors. File size  (complexity of the drawing) may have. My best guess is that something is seriously wrong with your network setup, so that packets randomly get lost and then you are be stuck until you restart both the lasercutter and VisiCut. The fact that you have more errors with larger files definitely points towards my guess.

1. Can you start VisiCut from the command line and post the error messages, so that we see the exact error?
https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline

2. Can you please double-check your network setup (try a new cable, try a direct connection to your laser cutter, try another network card / USB-Ethernet-Adapter)?

3. Can you please try to find some Windows machine (or setup a Windows VM) and test if your network and lasercutter setup works with the original Epilog driver? I understand that you want to use a Mac, but we must be sure that the problem is definitely with Visicut and not with your cutter or network setup, so we need something for comparison. Note that you don't need a Windows license, Windows 10 can be downloaded from Microsoft and, as far as I remember, works as a "demo version" for some weeks without entering a license key.

If it is clear that everything works with the Epilog driver, but Visicut fails from the same machine, then it we need a detailed recording of what's going on on the network, e.g. a `.pcap` file which cah be recorded with `tcpdump` (or some graphical program such as Wireshark). Only then we can compare the working behavior with what VisiCut does.

Sorry to bother you with all these questions, but it's not possible to help you without further information.

zebulong, 2019-05-05:
Just recently I tried again and didn’t get the error on some files, but on one I did.  I was still completely stumped and went to go back to the laser to delete the other files or restart it and I noticed there was a new file waiting, even tho Visicut have given the timeout error.  So I sent it again and went to the laser and waited- sure enough, the file came thru (and cut OK) but Visitcut said it had the timeout error.  So this is what I am finding, it actually does send it to the laser cutter, but it thinks it’s not!  At least that’s what where I am at with it now.    Should I still run the command line and post the error log?

Thank you for your help, you guys are amazing!
James

mgmax, 2019-05-05:
@zebulong Yes, please post more information so that we can help find the problem.
It may be something in VisiCut, or a network issue. I often experienced that a bad network cables make the network horribly slow, causing timeouts.

mgmax, 2019-11-01:
Closing the issue because the submitter did not respond for a longer time, and there is nothing we can do without a response. Feel free to respond with more information, so we can reopen and solve the issue.

zebulong, 2019-11-01:
Ok, but that is strange because I thought I replied.
The software is the part that is timing out, or at least saying it is, but the machine (Epilog Helix) is getting the file no problem, it turns out.

mgmax, 2019-11-02:
I received that reply, but we need more information to find the cause of the problem. It may as well be a network issue. Also, we sometimes had this error (on Epilog Zing, both with VisiCut and with the official Windows driver) if you press "X/Y off" and then do not set the start point before sending the job.

https://github.com/t-oster/VisiCut/issues/437#issuecomment-489442600

zebulong, 2019-11-02:
I am always sure to set the start point before send the job.
Not sure why it is happening.

mgmax, 2019-11-02:
Could you please try the steps I mentioned here, or at least some of them: https://github.com/t-oster/VisiCut/issues/437#issuecomment-489442600 This is what I would to to diagnose the issue. I would like to help but cannot do that myself without a Epilog Helix.

zebulong, 2019-11-04:
> Could you please try the steps I mentioned here, or at least some of them: #437 (comment) <https://github.com/t-oster/VisiCut/issues/437#issuecomment-489442600> This is what I would to to diagnose the issue. I would like to help but cannot do that myself without a Epilog Helix.

I would be glad to, but just today went to open the app and got the JRE Load Error.  I am trying to fix it, but cannot open any version of visicut at the moment…

mgmax, 2019-11-09:
sorry to hear that. Can you please open a new issue about that, including your OS and java version? (On Mac/Linux, you need to update to Java 9 or newer.)

zebulong, 2019-11-09:
I actually got that sorted out and soon I will do the test you requested

zebulong, 2019-11-09:
I will run this when I get the error next time.  I am not getting them now….

mgmax, 2019-12-31:
Closing the issue for now -- without detailed information (see https://github.com/t-oster/VisiCut/issues/437#issuecomment-489442600 )  I have to assume it's a network issue. Feel free to reply if you can reproduce it and provide details.

closed 2019-12-31 by mgmax

zebulong, 2019-12-31:
What happens is that Visicut says timeout error but actually, the file is being sent successfully to the mavhine after all. Visicut just doesnt think it is for some reason.

mgmax, 2019-12-31:
That may also be the result of a network problem (response is lost or 
delayed due to retransmissions).

zebulong, 2019-12-31:
Lets leave it closed still. I have a new computer and it hasnt happened since.