t-oster/VisiCut #438: file won't load in Visicut version 1.8-6 on Mac

zebulong, 2018-02-01:
1.8-5 works fine, but 1.8-6 will not load a file.  Both using the file button and dragging/dropping a file hangs up the progress bar and nothing loads.

mgmax, 2018-02-01:
I don't see any possible reason in the code. Can you try it on another Mac to make sure it's not special to your system?

zebulong, 2018-02-02:
oh, how interesting.  it loads on my other Mac!    any idea of what could be blocking the run of this app on my other mac machine???

I really like visicut- thanks for making it and if i can get the latest version to run, i’ll be even happier.   thanks if you have any suggestion.   maybe i should use app. delete to remove the older version???

james

hillct, 2018-02-04:
Is that an issue other than the documented GateKeeper / Application signature problem?

zebulong, 2018-02-05:
Yes, separate issue.    The program opens but it won't let me load a file.    But no worries, because the latest version that was just posted seems to work fine!        Thanks!

renebohne, 2018-04-29:
zebulong, so this ticket can be closed? Which version solved your problem?

zebulong, 2018-04-29:
1.8-11 seems to work ok, I just downloaded the latest one and it seems to be ok
Thanks!

closed 2018-07-04 by t-oster