t-oster/VisiCut #439: UI bug in 1.8.9 on Mac

hillct, 2018-02-04:
The attribute names fields (left column) contain no text, and do not appear editable (are they meant to be?) when using a template material as seen attached. When creating a new material from scratch, the field names are not editable either. These tests were performed on MacOS 10.13.
[image]
[image]
 java -version
java version "1.7.0_51"
Java(TM) SE Runtime Environment (build 1.7.0_51-b13)
Java HotSpot(TM) 64-Bit Server VM (build 24.51-b03, mixed mode)

mgmax, 2019-10-09:
We no longer support Java 7 ("1.7"). Please update to Java 8 or Java 11.

mgmax, 2019-11-01:
Closing because the report refers to a very old version. Feel free to reply if the bug also happens in a recent version of VisiCut with Java >= 9.

closed 2019-11-01 by mgmax