t-oster/VisiCut #444: Position reference point disregarded

zebulong, 2018-02-25:
Hello I am using the most current version of Visicut and am finding that the Position reference point is disregarded when I execute a file to the laser.  If I set it to the center point, the laser uses the top left default point.  I have triple checked what I know to do and still, even after restarting the program, it still disregards my assignment of the center position.  is this easily fixable?  thanks for your time!

t-oster, 2018-02-25:
Hi, which driver are you using?

zebulong, 2018-02-25:
Epilog Helix.    Is that what you mean?

t-oster, 2018-02-25:
Yes.The starting point feature is meant for the x/y-off manual head positioning of epilog.not sure If the Helix has it.

zebulong, 2018-02-25:
Yes, it does have that, but I do not understand how that would work with the software. If I want to manually position the laser headofthe helix, i press the x/y-off button on its control panel, movethehead with laser pointer to a different start location, then hit "set home".    The cutting then starts from here when visicut sends the job.    I thought the starting point feature of your software was to tell the laser what part of the job essentially is the red. Is that not how this works? I really would like to be able to make the starting point the center of my drawing, because it's so much easier to start to cut things on circular objects this way... thanks for your time!

mgmax, 2018-02-25:
@zebulong you understood it correctly. I added some documentation (and you are highly welcome to contribute.) https://github.com/t-oster/VisiCut/wiki/VisiCut-manual#positioning-resizing-and-rotating

It should not be too complicated to add more context menu entries, such as "set start point -> to drawing center". The code is around here:
https://github.com/t-oster/VisiCut/blob/029e56b1c2aa7858a1bcc630668be6d86c1717b4/src/com/t_oster/visicut/gui/PreviewPanelKeyboardMouseHandler.java#L114
https://github.com/t-oster/VisiCut/blob/029e56b1c2aa7858a1bcc630668be6d86c1717b4/src/com/t_oster/visicut/gui/PreviewPanelKeyboardMouseHandler.java#L241

Unfortunately, I don't have time for that.

t-oster, 2018-02-25:
Yes this is exactly how it should work. You position the red laser where the red starting point in the drawing should be. Beware of the boundaries because the laserhead might bump into them if there is not enough space

zebulong, 2018-02-25:
Ok, I think I get it.  The position reference point tab is just a measurement reference for where any of those points would be, but if I want the laser to originate in the center, I need to right-click and set the start point in the center.  Thanks for clarifying!

t-oster, 2018-02-26:
right. Feel free to update the wiki in order to help others.

closed 2018-02-26 by t-oster

zebulong, 2018-02-26:
I would, if i felt i had anything to add.  I think my issue was that i did not know that page existed!
now i do, i have saved the link to read there first before asking questions.  thanks again for your time and help!