t-oster/VisiCut #454: Facebook integration broken

renebohne, 2018-04-29:
I am using Mac OS X 10.12.3 with Java 1.8.0_131
flow: click on facebook in the menu, then a new browser window popped up, I granted access to VisiCut Connect and I got a long auth key in a new window. I copied this autho token to the VisiCut dialog and hit enter.
First error (why I see this dialog at all):  `JavaFX is not available. Using fallback behavior.`

And then the long error message, that might mean, that the facebook process changed or that the version of scribe is too old?
```
org.scribe.exceptions.OAuthException: Response body is incorrect. Can't extract a token from this: '{"access_token":"EAAHai0GCdj0BAFYC7kGxGgCobeQruakqOHpTxPkz3fZACt7dUP7HxZB3DYT1JOetEgAAd1kY6to8zTHecpZAGuTAnAERQwSNfKMYo6shBi7POVd9RIa86QJp6OMNDgNk3XF9oi4vHI6FMTrae02HouZBtyjJT28ZD","token_type":"bearer","expires_in":5183891}'
	at org.scribe.extractors.TokenExtractor20Impl.extract(TokenExtractor20Impl.java:32)
	at org.scribe.oauth.OAuth20ServiceImpl.getAccessToken(OAuth20ServiceImpl.java:39)
	at com.tur0kk.facebook.FacebookClient.loginWithBrowserCode(FacebookClient.java:105)
	at com.tur0kk.facebook.FacebookManager.logIn(FacebookManager.java:115)
	at com.t_oster.visicut.gui.MainView.startFacebook(MainView.java:3296)
```

mgmax, 2018-04-29:
Has anyone used the facebook / thingiverse / whatever integrations before, or can we just remove them?

t-oster, 2018-04-29:
Maybe we should think of a plugin architecture in order to keep such extensions separately from the main code...

commit 0dada19b (2020-01-01): Remove broken thingiverse and facebook integration

Closes #455
Closes #454
closed 2020-01-01 by mgmax