t-oster/VisiCut #455: Thingiverse integration is broken

renebohne, 2018-04-29:
I am using Mac OS X 10.12.3 with Java 1.8.0_131
flow: click on facebook in the menu, then a new browser window popped up, I granted access to VisiCut Connect and I got a long auth key in a new window. I copied this autho token to the VisiCut dialog and hit enter.
First error (why I see this dialog at all):  `JavaFX is not available. Using fallback behavior.`

And this is the long error message. Seems as if Thingiverse moved the URL for the API and we need to update the endpoint:

```
2018-04-29 15:05:20.584 JavaApplicationStub[41635:18411165] -_cgsEventRecord is deprecated for _cgsEventRecord. Please use -CGEvent.
TEST
Login failed! Response body is incorrect. Can't extract a token from this: '<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>301 Moved Permanently</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="https://www.thingiverse.com/login/oauth/access_token?client_id=a91afd857a7d6f2dba7f&amp;client_secret=00d763afa30cda6248fc203aa997eee2&amp;code=86ba4654381cf3007c92b19226472e42&amp;redirect_uri=http%3A%2F%2Fhci.rwth-aachen.de%2Fpublic%2FVisiCut%2Fshow_code.php">here</a>.</p>
<hr>
<address>Apache/2.4.7 (Ubuntu) Server at www.thingiverse.com Port 80</address>
</body></html>

```

t-oster, 2019-11-01:
Maybe we should remove the integration? Did anybody use it? Or we should create a plugin system, so that this can be developed independently?

renebohne, 2019-11-01:
We can remove it if it does not work.

A plugin system would be nice since sharing designs and showing results is
one of the main concepts of fab labs. Thingiverse is still the most
important platform for sharing design files.

commit 0dada19b (2020-01-01): Remove broken thingiverse and facebook integration

Closes #455
Closes #454
closed 2020-01-01 by mgmax