t-oster/VisiCut #458: UI scaling (vertical and horizontal) is buggy

mgmax, 2018-05-06:
No matter how large or small I resize the window, the scrollbars and UI resizing don't seem to make too much sense:

[image]

[image]

The problem might to be caused by the scroll box around the mapping/position/lasersettings tab pane. This scrollbox stubbornly refuses to adapt to the available space.

@mrlinux1 I would like to remove the scrollbox which you added in commit f11597ba462b721ad2776e5876b5b64ff7c64140 , or find another solution to fix the layout for large enough monitors. Do you have any opinion or suggestions?

jnweiger, 2018-05-07:
[image]

The intention of the scrollbox was to make it easier to reach the 'Execute' Button (Ausführen in German locale) in the lower right hand corner when screen space is limited. Currently the Scrollbar makes it possible to reach the Execute button without resizing. That is a good thing.

An enhancement would be (imho) to have a fixed footer with Calculate and Execute buttons always visible and a scrollbox for everything else above.

@mgmax Did you mean something different?

t-oster, 2018-05-28:
I did some changes to vertical resizing. Is this better: https://github.com/t-oster/VisiCut/pull/464

mgmax, 2018-08-18:
Now it's a lot better than before.

closed 2018-08-18 by mgmax