t-oster/VisiCut #466: Drv for V8.1 Motion Controller Card

mikeone531, 2018-06-20:
Hello,
I have a small Chinese cnc equipped with V8.1 Motion Controller Card for laser engraving and laser cutting machine is compatible with HPGL language. (http://www.rabbitlaserusa.com/DriverDisk/NewlyDraw/NewlyDraw_Driver_CD/Newlydraw_Driver_CD/%E8%AF%B4%E6%98%8E%E6%96%87%E4%BB%B6%E5%8F%8A%E5%9B%BE%E7%BA%B8/English%20specifications/V8.1%20Wiring%20Diagram.JPG)
I checked the documentation of your project and the documents on the net, but unfortunately I do not have enough knowledge to prepare the drivers or a correct configuration of "visicut".
From what I see on the net there are many who have machines with this card. Being able to drive the cnc with your sw would be great!

I hope you can help me, thanks in advance

mgmax, 2018-08-15:
If it speaks HPGL, try the "GoldCutHPGL" driver.
It may also help to "print to file" with the manufacturer supplied driver, look at the file and compare it with "Export Laser Code" from visicut.

mikeone531, 2019-05-25:
Hi!
Over the months, I have been doing tests and doing various researches.
However I could not interact with my cnc.
Is it possible that no one in the community has faced the problem of ST-3060GU cnc that seem to be quite common?
Is it possible that he is the first and only one?  ;-(

mgmax, 2020-01-01:
We cannot help with the little information that you gave. Can you post an example file that works? What exactly does not work with the HPGL produced by VisiCut GoldCutHPGL?

closed 2020-01-01 by mgmax

tatarize, 2023-02-21:
The newlydraw devices are not strictly HPGL2 standard, but a special laser variation of it. This is also transferred over the USB at the vid/pid 0x0471 0x0999.

The protocol is to connect to the WRITE_INTERRUPT endpoint transfer the packet size in int16le from 0000 to 1000 (4092) (Packets are truncated at 4092 and remainder is sent in next packet). Then read from the READ_INTERRUPT endpoint for the value 01 which means OK. Then send the packet of modified HPGL like code. This includes commands like UL, PR (HPGL), IN (HPGL), DW, ZED, SP1 (HPGL), SP2 (HPGL), BT8, TO#, DA#, PL#, BC#, BD#, TX#, PU (HPGL), PD (HPGL), ZZZFile#, VS#, BT#, VQ#, VP#, CV#, WV#, WU#, WR#, VX#, VY#, VM#, VJ#, VK#, CU#, WV#, WU#, WR#, CR#, PA (HPGL), ZQ, ZT, ZG, RS, ZH#, ZK#, ZG#, YF<scanline bits>, YZ <scanline bits>. 

See:
https://github.com/lynxis/laserusb

This is, strictly speaking an adaptation of the HPGL which doesn't have things like speed of the laser head, power of laser, corner compensation, or raster scanlines.

@mikeone531 if you're interesting there's a bit of a side project in Meerk40t to write a driver for it, but there's very few people with newlydraw boards. https://discord.gg/QGjkfEHUmC -- We're still mapping some stuff out and likely have a driver that might be able to jog. But, it depends on whether people are around that have newlydraw. And I currently we only have the one person. But, this is generally doable, though I never really got Java's usb stuff to work very well in windows which makes some of the work here a little harder.

But, generally this has a lot more than just running HPGL or whatever.