t-oster/VisiCut #468: Cannot load recommended settings under Linux Mint 19 (aka ubuntu 18.04)

jnweiger, 2018-07-31:
Using visicut_1.7-310-gcf8c087-1_all.deb from https://github.com/t-oster/VisiCut/releases (or my own builds, same issue):
Started with ~/.visicut removed
This is what I get, when loading the 'VisiCut example settings':
[image]

Nothing is loaded, and the console prints
```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.converters.collections.TreeMapConverter (file:/usr/share/visicut/lib/xstream-1.4.3.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.converters.collections.TreeMapConverter
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
'/home/testy/.visicut' doesn't exist. We create it.
Copying default settings...
done.
```
When trying to load RWTH Aachen (or Erlangen, or Nürnberg), I get this error:
[image]

and nothing is loaded.

@kkaempf suspected old xstream 1.3.2 to be related in https://github.com/fablabnbg/VisiCut/issues/31#issuecomment-408012625 -- I could not confirm this.

Using the system java openjdk-10 
When installing openjdk-8-jre and selected via `update-alternatives --config java`

Visicut does not run at all, it exits during startup:
```
rm -rf ~/.visicut
visicut 
 '/home/testy/.visicut' doesn't exist. We create it.
 Can't load settings.
```

Any other ideas?

jnweiger, 2018-07-31:
The very same systems also cannot run sikulisetup-1.1.2.jar from https://launchpad.net/sikuli/sikulix/1.1.2
- we fail with the same error message when trying to download the needed jar's. 

```
 java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors 
 parameter must be non-empty
 [error] RunSetup: Maven download: could not get timestamp nor buildnumber ...
```

I am probably missing something obvious on my machines. The issue is probably more related to Ubuntu 18.04 than to VisiCut,

t-oster, 2018-08-02:
@jnweiger thanks for finding that fix. I guess when downloading the settings from HTTPs it fails if the certificates are not up to date, because java cannot verify the server certificates then.

closed 2018-08-02 by t-oster