t-oster/VisiCut #473: Ubuntu 18, Visicut failing to launch

9Voltaire, 2018-08-23:
Hi there, been using Visicut for about a year. It's my bread and butter. Now suddenly it doesn't open .svgs or open independently. I'm running Ubuntu 18, which normally works with Visicut. I've uninstalled it and reinstalled it through all the recent versions. Sudo apt update and sudo apt upgrade have no effect.

When run through Terminal, I get the error code:

Exception in thread "main" java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.desktop/java.awt.Toolkit.newAWTError(Toolkit.java:472)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:488)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
	at java.base/java.util.HashMap$KeySpliterator.forEachRemaining(HashMap.java:1608)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at java.desktop/java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:532)
	at java.desktop/java.awt.Toolkit.getDefaultToolkit(Toolkit.java:613)
	at java.desktop/com.sun.java.swing.plaf.gtk.GTKLookAndFeel.isSupportedLookAndFeel(GTKLookAndFeel.java:1515)
	at java.desktop/javax.swing.UIManager.setLookAndFeel(UIManager.java:572)
	at java.desktop/javax.swing.UIManager.setLookAndFeel(UIManager.java:632)
	at com.t_oster.visicut.gui.VisicutApp.main(VisicutApp.java:143)
Caused by: java.lang.ClassNotFoundException: org.GNOME.Accessibility.AtkWrapper
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:374)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:485)
	... 15 more

I've purged openjdk and brought it back, having read it was a "headless" issue. I'm quite stuck, anyone have experience with this?

t-oster, 2018-08-24:
Hi, looks like a Bug in openjdk (https://bugs.launchpad.net/ubuntu/+source/openjdk-lts/+bug/1788250). Try to:

> Comment out line with 'assistive_technologies=org.GNOME.Accessibility.AtkWrapper'
in
/etc/java-11-openjdk/accessibility.properties

t-oster, 2018-08-24:
alternatively:
`export CLASSPATH=/usr/share/java/java-atk-wrapper.jar ` before starting VisiCut (https://bugs.openjdk.java.net/browse/JDK-8204862)

9Voltaire, 2018-08-24:
Neat! I tried commenting out that line and it's running now. Thank you so much!

closed 2018-08-24 by 9Voltaire