t-oster/VisiCut #477: null point exception in previously working file

LenoreHorner, 2018-09-21:
I successfully cut/engraved in plywood a particular file earlier this summer (same OS but earlier version of visicut).   Now when I open the file in Inkscape and use the extension open in visicut and then set up the plywood and what to engrave vs what to cut.  When I then try to send the file to the printer, sometimes I get a nullPointerException (addtolaserjob(vectorprofile.java:265) and visicutmodel.preparejob(visicutmodel.java:7gg))  [sorry if these aren't quite right; I'm can't copy from the error window.  But sometimes I get no error (all I did was x out of the error message and say cmd-print again) but then I get the "preparing job" forever.  Sometimes I have also gotten something that looks like a type mismatch error but I can't get it to do that one right now.

Note that since I'm on a mac, I'm using the cross-platform version.

mgmax, 2018-09-23:
I would like to help, but unfortunately I currently cannot say more than "sorry, it works for me".

Please try to figure out detailed instructions by which you can trigger the first error reliably (as far as this is possible). Screenshots of the error are also helpful.

If it is an extra message box saying something like "please restart visicut", the message will be shown only once until you start visicut, and any behavior after that error occured is completely random.

LenoreHorner, 2018-09-25:
This isn't happening for me on either machine I was working on before, so I think it's a moot point.

closed 2018-09-25 by LenoreHorner