t-oster/VisiCut #478: IllegalFormatConversionException

LenoreHorner, 2018-09-21:
The other error I'm getting - and I don't know whether it's a bug in visicut itself or in the extension that extracts the file from Inkscape and converts text - is IllegalFormatConversionException: d!=java.lang.Double

LenoreHorner, 2018-09-21:
I tried having inkscape create a spiral with the spiral tool and then used the extension to send the drawing to visicut and got the error above so it's not just text that's causing this.

t-oster, 2018-09-22:
Could you try older versions and find the first which doesnt work?

mgmax, 2018-09-23:
@t-oster could you please update the LibLaserCut version in VisiCut? I hope this bug will be solved by https://github.com/t-oster/LibLaserCut/pull/93

t-oster, 2018-09-24:
should be fixed. Please reoen, if the issue persists.

closed 2018-09-24 by t-oster