t-oster/VisiCut #480: Use SVG support with LibLaserCut

WieFel, 2018-09-24:
Hello,
I am trying to develop a tool which takes an SVG file as an input and sends it to the laser cutter as a job. I am using LibLaserCut, which doesn't offer SVGs to be read and sent to the cutter.
Therefore, I am trying to use the tools from _visicut.model.graphicelements_ to import an SVG file.
However, I have difficulties doing so. I cannot clearly find the lines where this is done in the VisiCut code.

Do you have suggestions or a small example on how I could use the classes of _visicut.model.graphicelements_ to import an SVG and send it as a job?

t-oster, 2018-09-25:
Well the SVG itself does not contain any information on how to cut/engrave what parts. That's what VisiCut calls a mapping.
Maybe it is easier for you to just use VisiCut to create your Mapping and Lasercutter settings and then call VisiCut from the command line (or include the jar and manually run the main method). It is possible to use VisiCut without GUI.

t-oster, 2018-09-25:
https://github.com/t-oster/VisiCut/blob/master/src/com/t_oster/visicut/gui/VisicutApp.java#L269

closed 2019-01-31 by t-oster