t-oster/VisiCut #481: can engrave but not cut

LenoreHorner, 2018-09-25:
If I create text inclosed in an ellipse in Inkscape and use the extension to open that in VisiCut, then if I just engrave the text, things work fine, but if I try to cut around the text or even just cut the ellipse, Visicut just thinks forever.  MacOS never sees it as "not responding" but nothing happens either.  See screen shots.  I tried thinning the ellipse border worried that VisiCut was trying to cut too wide a line but at 0.1mm thickness, I still get hanging.  
[image]
[image]
[image]

mgmax, 2018-09-25:
Are you using the latest version? Do you get any output when running from the commandline?
In case you don't know how to run from commandline, I just created a wiki article https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline

LenoreHorner, 2018-09-25:
I made just the outline of an ellipse in Inkscape, started VisiCut from the command line (it's ./VisiCut.MacOS really) and imported the .svg of the ellipse.  I chose 4mm plywood, cut all, and set power and speed according to the Helix manual.  The output I get on the command line follows.
2018-09-25 14:24:45.073 java[9273:2758420] plugin com.adobe.accmac.ACCFinderSync invalidated
Exception in thread "Thread-5" java.lang.NoSuchMethodError: com.t_oster.liblasercut.VectorCommand.getX()I
	at com.t_oster.liblasercut.vectoroptimizers.VectorOptimizer.divide(VectorOptimizer.java:181)
	at com.t_oster.liblasercut.vectoroptimizers.VectorOptimizer.optimize(VectorOptimizer.java:225)
	at com.t_oster.visicut.model.VectorProfile.addToLaserJob(VectorProfile.java:311)
	at com.t_oster.visicut.VisicutModel.prepareJob(VisicutModel.java:766)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:782)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2255)

mgmax, 2018-09-25:
ah okay, this is bug #479. Can you try downloading the newest version again? Until recently the download had exactly this error, it was replaced by a corrected upload (with the same filename and version number).

LenoreHorner, 2018-09-26:
I have downloaded the newest version and that does fix the problem.  Thank you.

closed 2018-09-26 by t-oster