t-oster/VisiCut #482: null pointer iff contains text

LenoreHorner, 2018-10-03:
Using 1.8-51 (cross-platform being used on MacOS 10.13.6) I can engrave text and cut around it at least sometimes.  However, with a particular file (which I can't attach b/c its svg) composed of two imported svg files plus some other shapes and text created in Inkscape, using the extension in Inkscape to send to VisiCut and then attempting to send the file to the Helix gives the attached error. 
[image]
  I tried a few other fonts with no change in behavior.  I removed the text entirely and could send the file and have the engraving and cutting done correctly.  I could then send just the text part and have it engrave correctly as well.

t-oster, 2018-10-04:
Well the exception your screenshot shows looks like as is has to do with the mapping and not with the file. Please upload the SVG somewhere and post a link and describe what exactly you click before the error occurs.
Also please try to start VisiCut from the command line in oder to get a more detailed error output.

LenoreHorner, 2018-10-04:
svg file at https://www.dropbox.com/s/tgwxn52wi7ky8a7/WholeDesign.svg?dl=0
steps:
in inkscape
1. extensions -> lasercut path -> open in visicut
in visicut
2. material -> birkensperrholz(multiplex)
3. material thickness + -> set value to 6 mm
4. mapping -> map by single property -> color -> choose as shown in image
5. laser settings -> set power and speed settings for cut and for engrave according to Helix manual (sorry; don't have numbers memorized and I'm not at school again until Tuesday)
6. plug in both ends of usb->ethernet cable and turn on helix
7. cmd-P

I finally managed to get the fonts turned to paths in Inkscape (though I get a message that says there was nothing to do).  Here's what starting from the command line gives.
objc[75544]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffb1586c90) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x1238adcd8). One of the two will be used. Which one is undefined.
java.lang.NullPointerException
	at com.t_oster.visicut.model.RasterProfile.addToLaserJob(RasterProfile.java:240)
	at com.t_oster.visicut.VisicutModel.prepareJob(VisicutModel.java:766)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:782)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2255)
java.lang.NullPointerException
	at com.t_oster.visicut.model.RasterProfile.addToLaserJob(RasterProfile.java:240)
	at com.t_oster.visicut.VisicutModel.prepareJob(VisicutModel.java:766)
	at com.t_oster.visicut.VisicutModel.sendJob(VisicutModel.java:782)
	at com.t_oster.visicut.gui.MainView$65.run(MainView.java:2255)

mgmax, 2018-10-04:
I don't think the problem is with text, but with the default DPI for the "engrave" profile.

@LenoreHorner can you try changing Options -> Edit Laserprofiles -> engrave -> set DPI to 400, restart visicut and see if it works now?

Here is how to reproduce the error without any special file:
- download fablab aachen settings
- examples -> giveaway_phoneStand.svg
- material Baumwolle (doesn't matter)
- add epilog helix cutter (no settings required, just click save) and select it
- map by stroke color
- red -> engrave, green -> engrave
- execute
- message box "500dpi is not supported. use 400dpi?" -> yes
- error message
- press execute again and everything seems to work(?), at least there is no error

My best guess is that the "unsupported DPI workaround" in 
https://github.com/t-oster/VisiCut/blob/0beb401ea0c0f78961e9f1db86fc8cab7cf5524e/src/com/t_oster/visicut/gui/MainView.java#L3672
fails as soon as there are two or more colors mapped to the same profile. It seems that the mapping representation does not map to the same instance, but to equal copies, which means that in the mapping we change only the first instance's DPI, and the following ones (second/third/... engrave color) now point to a laserprofile that has no properties.

The best solution would be to move this workaround to prepareJob() here and only change a copy of the profile `pr`:
https://github.com/t-oster/VisiCut/blob/af8129d9a46fc7ca15ef51a57dbeab3f25de6510/src/com/t_oster/visicut/VisicutModel.java#L766

LenoreHorner, 2018-10-05:
I'm not even in the same town as the Helix at the moment, so I can't do a full check.  I changed the settings to 400 as instructed and now all I get is a connection error (for obvious reasons) so probably that fixed my problem.  I'd been hunting around for where the 500dpi was coming from.  I've set 600 various places and it never seemed to stick.  Now I know why.