t-oster/VisiCut #484: Downloads fail

peteruithoven, 2018-10-26:
I and a friend can't download Visicut releases on https://download.visicut.org/ at the moment. 
Both in Firefox and Chrome. OS doesn't seem to matter either. 
It seems to fail at the last moment. 

I've been able to download a version successfully on 13 oct.

t-oster, 2018-10-26:
I updated my server config, will look into it tomorrow

t-oster, 2018-10-27:
It should be fixed/worked around now. I was giving a header `Content-Length` to tell the browser how big the downloaded file is, so it can output a percentage. However, since the file is compressed during the transfer by my webserver/proxy, the length I am giving does not match the length of the compressed stream, which causes the browser to assume an error when the stream ends.

I just removed that header, so the file will be downloaded correctly. But you won't see any progress bar/percentage anymore...

closed 2018-10-27 by t-oster

peteruithoven, 2018-10-27:
Damn, thanks for fixing it so quickly.