t-oster/VisiCut #485: advanced mapping selection filter error on Windows

grantgibson45, 2018-11-01:
There is an issue in the Windows version (1.8.21.1) - Windows 10 Home 64 bit build 17134 Java version 8 191 

The advanced mapping selection filters are greyed out such that you cannot choose a colour or setting to add or modify the existing setting. The tree is collapsed and cannot be opened. 

[image]

alastair-dm, 2018-11-06:
More investigation needed - the working machine is also running Windows 10 Home 10.0.17134 and Java Version 8 Update 191 build 1.8.0_191-b12.

alastair-dm, 2018-11-26:
Grant - please double check operation with a file that definitely has more that one identifiable characteristic. An SVG of 2 shapes with different stoke colour should do.

mgmax, 2019-11-01:
Closing the issue because the submitter did not respond for a longer time, and there is nothing we can do without a response. Feel free to respond with more information, so we can reopen and solve the issue.

closed 2019-11-01 by mgmax