t-oster/VisiCut #487: exporting bounding boxes

LenoreHorner, 2018-12-03:
When I use the Inkscape extension to send and svg image (made up of several copies of an original image) the drawing and the bounding boxes get exported with the same color to VisiCut (no matter whether I have each member selected or nothing selected).  This naturally makes the file not work out at all well.  Why are the bounding boxes being treated as lines that I drew?
[image]
[image]

LenoreHorner, 2018-12-03:
Using group so that there is only one bounding box when selected in Inkscape changes nothing.  Do I have to hand-edit the original svg file so it doesn't have a bounding box?

LenoreHorner, 2018-12-04:
Thanks. Hadn't thought of checking using a web browser.  Looks fine there.  File is at https://www.dropbox.com/s/avqd5h6fhu03hel/Koch2-Compound.svg?dl=0.

mgmax, 2018-12-05:
Your file contains <svg:clipPath> elements which are currently unsupported. See also #15.
The bug is that Visicut should ignore them, but doesn't.

LenoreHorner, 2018-12-05:
Thank you!  That enables me to fix my files by hand as well as to take this up with the writers of the software that drew everything via clipPath elements.

mgmax, 2019-11-01:
Closing - duplicate of #15

closed 2019-11-01 by mgmax