t-oster/VisiCut #492: consider listing features more prominently in Readme

matkoniecz, 2018-12-26:
https://github.com/t-oster/VisiCut#visicut - maybe screenshot or two to demonstrate that it is userfriendly?

First feature that is actually described is "upload and post a picture of a thing you made to Facebook", what - at least to me - is not something actually useful.

mgmax, 2018-12-26:
Thanks for pointing this out - I agree, and help with this would be appreciated. A list of important features could be copied from the website https://visicut.org (the sourcecode of the website is in the `gh-pages` branch). Adding screenshots to the website would be great -- and in my opinion the facebook/thingiverse stuff could be moved to the manual in the Wiki.

t-oster, 2019-01-31:
Yes, I am with you. Please help. Maybe we even remove the facebook and thingiverse stuff if no one needs it. On the other hand, as long as it does not cause issues, we can leave it there but it should not be listed as the first feature I agree