t-oster/VisiCut #497: latest builds are giving errors and/or not opening

zebulong, 2018-12-30:
I am running Mac OS 10.14.2 and when I open Visit (latest build) 1.8-58  I am getting various errors and then the app crashes.  a few versions older will load and run, but still getting error screens at the start.

thanks for your help!

here are the screen shots of the errors:
[image]
[image]
[image]

t-oster, 2019-01-16:
Seems to be another Java 10 with OSX issue like #488.
I need access to a Mac in order to test replacing the old "make java look good on OS X" stuff with java 10 compatible versions. Maybe @frattay can help me out...

zebulong, 2019-01-16:
Thanks for doing it when you get ahold of one and have time, many of us Mac folks appreciate it!

amiliv72, 2019-01-21:
Thanks for looking into it. While it is being fixed, it would be awesome if download page contained the last known good version of VisiCut for OS X. I'm kind of stuck currently, on a new Mac, don't have old version that worked, and the download page has only the last 4 versions, all of which are affected.

amiliv72, 2019-01-21:
I managed to find https://github.com/t-oster/VisiCut/releases that contains 1.7_310 release. Looks it should work.

t-oster, 2019-01-22:
Its probably not the VisiCut Version,but the Java Version which is to blame. Can you try with java8?

amiliv72, 2019-01-22:
I've seen people having this same kind of issue with all kinds of both open source and commercial Java applications on Macs (e.g. OpenRocket, one of my other favorite apps), where installed version(s) of Java and what the app actually needs differ. Would it be possible to package the app with the correct Java version? I've seen people doing it, never done it myself, but apparently it's just running a single command. Basically, what this dude did here for OpenRocket app in order to solve the same issue for all of us that are into rocketry: https://www.rocketryforum.com/threads/openrocket-packaged-installers-for-windows-and-mac-to-solve-all-your-java-problems.143540/

t-oster, 2019-01-22:
Yes,thats what I wanted to do for a Long Time, just ship the correct jre with the App. and then possibly sign it and add it to the App store.but i couldnt find any willing and capable osx Developer yet. However I managed to get OSX running in virtualbox so If I get time I can do it myself. Thanks for the Link.

t-oster, 2019-01-29:
Hi, the latest (1.8.62) OS X build should work (although you need to have oracle java 8 installed from their website).
I also found out how to include the JRE into the app bundle so you would not have to have java installed. However this will bloat the download by 200-300MB and I have to check legal issues.

t-oster, 2019-01-31:
@zebulong please confirm if those issues are resolved and then close this.

zebulong, 2019-01-31:
sorry, i have been away from home to do a family memorial service on the other side of the country and will be still another week here.  its a 2 week trip- so i havent had time to run any software check.  i do not even have a computer with me that runs software, just my phone.  go ahead and close it if you want, sounds like it's fixed....

t-oster, 2019-02-01:
Please reopen if not fixed

closed 2019-02-01 by t-oster

zebulong, 2019-05-21:
I am trying to, but when I open Terminal and type "visicut" like the directions above say, I get "command not found" error.

t-oster, 2019-05-22:
@zebulong if you are using the platform independent version, you have to run `./VisiCut.MacOS` but please try the regular OSX Verison from https://download.visicut.org/.
Just run `java -v` in the terminal in order to confirm that you are using java 1.8 and not java 10