t-oster/VisiCut #502: Empty "Exemple" folder in the MacOS version

waumand, 2019-03-24:
Hi makers !
With last Java 8 (version 201) with MacOS Mojave and 3 different Mac I can say the "Exemple" folder is empty.

I can't use exemples to make some shapes :/

Any idea ?

[image]
[image]

t-oster, 2019-03-24:
Wich settings did you download?

waumand, 2019-03-24:
@t-oster when you say setting you speak about setting file ?
I use this one for my Fablab with the RS-1060L 3D printer
[visicut-1060.vcsettings.zip](https://github.com/t-oster/VisiCut/files/3000818/visicut-1060.vcsettings.zip)
http://robotseed.com/index.php?id_category=11&controller=category&id_lang=1

t-oster, 2019-03-26:
I can confirm this. The examples included in the specific downloaded settings work, but the predefined examples are not found. Probably a path issue. I won't have time in the next two week though, but after that I will look into it.

closed 2021-10-17 by t-oster